m5stack/M5Stack #14: How to support for PlatformIO?

sczh01, 2017-09-05:
Hi Team,

Would we support PlatformIO?

thanks
tony

imxieyi, 2017-09-05:
I think you can refer to the guide on [arduino-esp32](https://github.com/espressif/arduino-esp32)

avaldebe, 2017-09-12:
Maybe I'm misunderstanding the question.

The m5stack is already supported by PlatformIO. I'm running PlatformIO Atom plugin (core 3.5.0a9). Just started a new project for the `m5stack-core-esp32` from the IDE and got the following entry to my `platformio.ini`. 

```
[env:m5stack-core-esp32]
platform = espressif32
board = m5stack-core-esp32
framework = arduino
```
After that all I needed to do was to put this library on the `lib/` directory, and could compile the examples

sczh01, 2017-09-13:
.thanks it's my request in platformio.

closed 2017-09-27 by m5stack