m5stack/M5Stack #16: How to confirm a faulty display?

avaldebe, 2017-09-12:
Hi,
I got my m5stack core today (nice presentation, kudos) and after I plugged in all I got was a white screen.
To test the unit I set up a project on PlatformIO for the `m5stack-core-esp32` and compiled the basic examples without any issues. The `Buttons.ino` and `Speaker.ino` worked fine, but `Display.ino` and `Hello.ino` showed nothing but a white screen.

I suspect this is a hardware issue. Can I get some help to confirm my suspicion?

avaldebe, 2017-09-12:
I adapted some of the diagnostics for one of the Adafruit_ILI9341 examples

```Arduino
#include <M5Stack.h>

// the setup routine runs once when M5Stack starts up
void setup(){

  // Initialize the M5Stack object
  m5.begin();

  // LCD display
  m5.Lcd.printf("hello world");

  Serial.printf("\r\nILI9341 Diagnostics\r\n");
  Serial.printf("Display Power Mode: 0x%02X\r\n",m5.Lcd.readcommand8(ILI9341_RDMODE));
  Serial.printf("MADCTL Mode:        0x%02X\r\n",m5.Lcd.readcommand8(ILI9341_RDMADCTL));
  Serial.printf("Pixel Format:       0x%02X\r\n",m5.Lcd.readcommand8(ILI9341_RDPIXFMT));
  Serial.printf("Image Format:       0x%02X\r\n",m5.Lcd.readcommand8(ILI9341_RDIMGFMT));
  Serial.printf("Self Diagnostic:    0x%02X\r\n",m5.Lcd.readcommand8(ILI9341_RDSELFDIAG));
}

// the loop routine runs over and over again forever
void loop() {

}
```
and get
```
[...]
M5Stack init...

ILI9341 Diagnostics
Display Power Mode: 0xFF
MADCTL Mode:        0xFF
Pixel Format:       0xFF
Image Format:       0xFF
Self Diagnostic:    0xFF
```

m5stack, 2017-09-13:
> m5stack/M5Stack <M5Stack@noreply.github.com>抄　送：Subscribed <subscribed@noreply.github.com>主　题：[m5stack/M5Stack] How to confirm a faulty display? (#16)

Hi Alvaro,   So sorry! I think the screen is broken, I will ship a new core to you this morning.:-)Thank you!

m5stack, 2017-09-13:
Please tell me your address and phone number, i cann't find your data from

avaldebe, 2017-09-13:
I'm afraid I did not get my unit from you directly. I'll send all the details and you my contact info via m5stack.com

closed 2017-09-27 by m5stack