m5stack/M5Stack #26: Failed to connect to ESP32

Smrtokvitek, 2017-11-17:
I am not able to upload the program to M5Stack via USB. I got this error:
`Failed to connect to ESP32: Timed out waiting for packet header`
I have win10, Arduino 1.6.12 and I have tried different drivers for CPX10X (6.7 and 6.7.5) without any change. I have also tried different CPU speeds in the settings, different baud rates for upload and different board settings. Nothing worked. It seems like `esptool.py` can't get the communication working. I can see/hear M5Stack reseting while esptool is trying to upload the program. I have 2 of these boards and they act the same so it shouldn't be a hardware issue.
I can see the output from the default application in my serial monitor without any problem.
The same problem happens in platformIO.

alanroche2015, 2017-11-19:
I had the same problem using VMware fusion with windows 10.  Had to use the Mac directly to get it to work.  Looks like a delay getting the serial data to the VM.

0x1abin, 2017-11-20:
hi, reference: 
https://github.com/espressif/arduino-esp32/issues/333#issuecomment-303515054
uninstall the usb driver ,try the low version.

Smrtokvitek, 2017-11-21:
I tried different PC and it works even with a default win10 driver (v6.7). No driver installation needed. Straight out of a box. Same version of Arduino and M5Stack libs. So it looks like some issue with my other PC. :-/

closed 2018-09-05 by sakabin