m5stack/M5Stack #27: Screen Flicker and click sound when flashing

tobozo, 2017-11-24:
Hello :-)

I've tried two different systems (Ubuntu and Win10) so far, where I've successfully installed the drivers and get the M5Stack detected in Arduino 1.8.5.

On both systems when I hit the upload button in the Arduino IDE, the screen on the M5Stack starts flickering and I can hear a click sound.

After a while I get a timeout error.

```
esptool.py v2.1
Connecting........_____....._____....._____....._____....._____....._____.....
Traceback (most recent call last):
  File "esptool.py", line 2524, in <module>
  File "esptool.py", line 2517, in _main
  File "esptool.py", line 2247, in main
  File "esptool.py", line 368, in connect
  File "esptool.py", line 344, in _connect_attempt
  File "esptool.py", line 297, in sync
  File "esptool.py", line 244, in command
  File "esptool.py", line 227, in write
  File "site-packages\serial\serialwin32.py", line 323, in write
serial.serialutil.SerialTimeoutException: Write timeout
Failed to execute script esptool
```

Also I noticed that while the cable is a USB-C, it'll only work on one side, how symmetrical is that, and how can I make sure this cable isn't faulty (I don't have any spares)?

tobozo, 2017-11-25:
I did some more tests and it doesn't seem to be the cable.

Using the same cp210x USB UART I use to flash other MCUs, I connected the pins and got the serial output all right, but still the same reset loop while attempting to flash.

Tried it on another Windows10 machine, did a full install of Arduino IDE 1.8.5 + ESP32 SDK install + M5Stack Core environment install, and same reset loop while flashing.

Could it be that I have a faulty unit?

tobozo, 2017-11-25:
Quick fix by implementing manual flash button (ghetto style) between GPIO0 and GND.
 
[image]

closed 2017-11-25 by tobozo

MarsTechHAN, 2017-11-26:
We suspect that as it tries to pull down GPIO0 and restart the ESP32, it triggered the "shutdown" function of the PMIC, and failed to meet the timespan that esptool think the esp32 has already booted. We thought it could be possible when we initially design the circuit, but the test results show it seldom happened, and in order to get rid of it would yield much more complex design which we have no space to put(as you can see :(   ). If it happened all the time, there must be somewhere wrong, we will conduct more test and try to fix it in the next version. (Sorry for your poor experience  : O

tobozo, 2017-11-26:
Happy to help :-) 

I tried with other cp210x based devices that used to work fine (without requiring a driver update) like [this one](https://www.tindie.com/products/fred_IOT/esp32s-esp-wroom32-frogopins-development-board/).

Since the device/manufacturer ID's are similar to the one on the M5Stack and flashing works fine on those devices, can I assume I I'm using the correct version of the drivers?

M5Stack device/manufacturer IDs
```
Nov 26 12:19:02 kernel: [224098.163931] usb 6-2: New USB device found, idVendor=10c4, idProduct=ea60
Nov 26 12:19:02 kernel: [224098.163936] usb 6-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 26 12:19:02 kernel: [224098.163940] usb 6-2: Product: CP2104 USB to UART Bridge Controller
Nov 26 12:19:02 kernel: [224098.163942] usb 6-2: Manufacturer: Silicon Labs
Nov 26 12:19:02 kernel: [224098.163945] usb 6-2: SerialNumber: 015E6FE0
```

ESP32 Dev Board  device/manufacturer IDs
```
Nov 26 12:57:35 kernel: [226411.793687] usb 7-1: New USB device found, idVendor=10c4, idProduct=ea60
Nov 26 12:57:35 kernel: [226411.793693] usb 7-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Nov 26 12:57:35 kernel: [226411.793696] usb 7-1: Product: CP2104 USB to UART Bridge Controller
Nov 26 12:57:35 kernel: [226411.793699] usb 7-1: Manufacturer: Silicon Labs
Nov 26 12:57:35 kernel: [226411.793701] usb 7-1: SerialNumber: 015313B3

```

cbrightly, 2017-11-27:
I to experience this issue on a regular basis and have found that shorting ground to the reset pin temporarily will occasionally resolve this issue

tobozo, 2017-11-27:
I'll probably print [this](https://github.com/m5stack/M5-hardware/blob/master/PROTO.stl) and glue a button, OTA updates are not that stable yet and it adds a limit to sketch size.

dsprogis, 2018-01-17:
I am not clear on the recommendation here.  I ran a jumper from GPIO|G5 to GND - see photo:
[image]

I am still getting the same error:

esptool.py v2.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____.....____An error occurred while uploading the sketch
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

tobozo, 2018-01-17:
From the [latest announcement about this issue](https://twitter.com/M5Stack/status/950699238127644672), it's suggested to use a 2.2uf capacitor between RST and GND.