m5stack/M5Stack #28: BMP to Array

alanroche2015, 2017-11-27:
Hi 

Does anyone have documentation on how to convert a BMP to an array that works with the M5Stack ? I have tried a few methods but without success.

Thanks

vvs551, 2017-12-05:
Hi,
I usually do it with http://www.riuson.com/lcd-image-converter
You should save bmp in 8-bit format for use it with converter

Yazzcat, 2017-12-06:
I'm looking for the same possibility to convert an image.

I'v tried your tool, but have no clue what settings for image conversion I need... I can get it to compile, but it only shows garbage on the screen, whatever I try ;-)

Yazzcat, 2017-12-09:
Ah. Got it! Use the following settings:
Preset: R5G6B5
Block size: 16bit
Byte order: Big-Endian
Leave the rest as it is.

You will need to delete the 'Static' from the .c file and make the array externally available using the extern statement in your code.

Works beautifully!

Zontex, 2020-11-11:
Glad the problem got solved :) closing the issue.

closed 2020-11-11 by Zontex