m5stack/M5Stack #29: Power the M5Stack from the GPIO Ports

alanroche2015, 2017-11-27:
Is it possible to power the M5Stack from a GPIO PIN ? i don't want to use the USB port if possible.

0x1abin, 2017-11-29:
hi, 
https://github.com/m5stack/M5Stack/blob/master/examples/Basics/PowerOFF/PowerOFF.ino

alanroche2015, 2017-11-29:
Thanks for getting back to me, but I was hoping to power the device from GPIO PIN directly. Is it possible to apply 5 volts of 3.7 volts to one of the PIN’s to power the device.  

Thanks

tuupola, 2017-11-30:
Connecting power to GPIO pins will most likely burn your board. I believe you could use the BAT pin though.

Zontex, 2020-11-11:
Hello, you can power the M5Stack from the 5V pin directly, make sure it's 5V pin and not any other pin or it will burn your board and cause damage. you can check if it's 5V pin by using multimeter by plugging the board by USB cable first and see which pin has 5V output, then disconnect the USB and use the same pin for VIN 5V to power your board.
We don't support this way officially, do it at your own risk.

closed 2020-11-11 by Zontex

jnawk, 2023-05-20:
That suggests that in this diagram, the 5V pin is connected to either VIN or VIN_5V 

The difference being there is L2 between VIN and VIN_5V (and D1 after VIN_5V)

[image]

https://static-cdn.m5stack.com/resource/docs/products/core/stamp_c3u/stamp_c3u_sch_01.webp