m5stack/M5Stack #30: Hardware reset error

DeltaCore, 2017-11-30:
I have noticed an error in the current schematic.

At the transistor VT3: Collector and Emitter are switched.
[image]

This error prevents the usb reset circuit to work properly.

A easy workaround was to manually bridge GPIO0 and GND to flash a OTA firmware.

Best regards
 - Marcel / deltacore

MarsTechHAN, 2017-12-07:
Sorry for misleading, we notice the schematics we put on were a very old version. We will replace it as soon as possible. 
This image is from the PCB file that we send to manufaturing. 
[image]
And the new version of pdf.
[image]

DeltaCore, 2017-12-09:
This was my solution :P

[image]

I turned the transistor on it's head and rotated it so the emitter and collector are switched.

Best regards
 - Marcel

DeltaCore, 2017-12-10:
To be clear, i was not able to program the m5stack without the soldering i did my self.
I will probe the connections on my board to check if i have the board you attached a picture of or a different one.

Best regards
 - Marcel

closed 2020-11-12 by Zontex