m5stack/M5Stack #33: Flashing problem

reaper7, 2017-12-27:
Today I received my m5stack (limited version)
I can't flash any new sketch to this board :(
I still get this (windows 10 (64); Arduino 1.8.5; esp32 latest from github :
```
esptool.py v2.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____.....____Podczas ładowania szkicu wystąpił błąd
_

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
```
I try different version of drivers:
6.7.0, 6.7.5 with and without enumeration, 10.1.1. and old 4.40 ... still nothing!

but my other esp boards with CP210x uart2usb works with all above drivers ok.

Someone and especially the manufacturer have a solution for this problem??

srolf, 2017-12-28:
Please have a look at this workaround: [http://forum.m5stack.com/topic/55/simple-fix-when-upload-fails](http://forum.m5stack.com/topic/55/simple-fix-when-upload-fails
)
In order to deal with this issue, I created a mini bread board adapter which includes also the mentioned capacitor. My observation without the capacitor were pretty the same as in #27

reaper7, 2017-12-28:
I did two things, one is capacitor soldered inside (near 2x15 bus)
and second is transitor from @DeltaCore comment:
https://github.com/m5stack/M5Stack/issues/30#issuecomment-350450274 

now is ok

srolf, 2017-12-28:
The transistor was not causing a problem in my case. Is there more than one revision existing?

Netoperz, 2018-10-01:
I have the same problem as reaper.

Those boards are not so cheap, and i see that they are buggy as hell in terms of hardware design.
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
How to flash this device using arduino ide on windows 10 ?

Netoperz, 2018-10-01:
When i pressed the reset button and hold it 6 seconds, than pressed flash in arduino, and when it wanted to flash i have let go the button (red on M5) it started flashing
esptool.py v2.3.1
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 16384.2 kbit/s)...
Hash of data verified.
Compressed 15072 bytes to 9733...

Writing at 0x00001000... (100 %)
Wrote 15072 bytes (9733 compressed) at 0x00001000 in 0.1 seconds (effective 831.6 kbit/s)...
Hash of data verified.
Compressed 296880 bytes to 132593...

Writing at 0x00010000... (11 %)
Writing at 0x00014000... (22 %)
Writing at 0x00018000... (33 %)
Writing at 0x0001c000... (44 %)
Writing at 0x00020000... (55 %)
Writing at 0x00024000... (66 %)
Writing at 0x00028000... (77 %)
Writing at 0x0002c000... (88 %)
Writing at 0x00030000... (100 %)
Wrote 296880 bytes (132593 compressed) at 0x00010000 in 2.6 seconds (effective 907.9 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 3072.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

Vey very strange, the same with tooll, sometimes it works sometimes it does not, i have 8 M5's  Core  , gray and black

Netoperz, 2018-10-01:
I have changed the upload speed to 115200 and installed  CP210x Windows driver from silabs (64bit in my case) 
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers
And it works for now.

DarknessFX, 2019-07-27:
> I have changed the upload speed to 115200 and installed CP210x Windows driver from silabs (64bit in my case)
> https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers
> And it works for now.

This fixed my flashing problem, thanks.

chegewara, 2019-08-06:
Im having even worse issue with my m5stack fire. It was working for some time, but now i cant even to erase it with `make erase_flash`. Here is log:
```
Erasing entire flash...
esptool.py v2.6
Serial port /dev/ttyUSB0
Connecting.......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 84:0d:8e:1d:df:90
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Erasing flash (this may take a while)...
Chip erase completed successfully in 0.0s
Hard resetting via RTS pin...
```
It looks good, like everything works, except one thing, flash is not erased at all.

Zontex, 2020-11-12:
Changing upload speed / updating the driver for CP210x should solve the problem, hope it can help.

closed 2020-11-12 by Zontex