m5stack/M5Stack #34: M5 Deepsleep not waking up

alanroche2015, 2018-01-06:
Hi all

Testing deep sleep on the m5 and it appears that once you enter deepsleep you have to leave the usb cable connected to get it wake up again. Anyway to avoid having to leave the usb cable connected ?

alanroche2015, 2018-01-06:
I also see a battery connector on the m5 board, if I use this connector will the m5 be permanently powered ?

m5stack, 2018-01-06:
sorry, we need to Make an experiment about this.

alanroche2015, 2018-01-06:
Can be tested easily, disconnect the usb power, tell the m5 to enter deepsleep for 10 minutes and wake up. After the 10 minutes have passed the m5 will not wake up, if you leave the usb cable connected it wakes up perfectly.

stroobandt, 2018-03-05:
I can confirm this. I am experiencing exactly the same issue.
There is even no need to wait 10 minutes; wakeup on battery fails even after 2 minutes of deepsleep.

stroobandt, 2018-03-05:
This is a hardware problem of the IP5306 5V buck converter going into standby after 32 seconds when the load current remains below 45mA.
It is explained in this [M5Stack thread](http://forum.m5stack.com/topic/62/ip5306-automatic-standby)

coelner, 2018-05-11:
@m5stack  Is the problem solved?

Fuellhase, 2018-11-23:
Is the problem “No wakeup from deepsleep without USB connected“ solved in the meantime?

stroobandt, 2018-11-23:
No, it is not. As a matter of fact, **I totally do not recommend buying M5 devices.**

The black plastic of the two devices I got, became brittle after a month or two.
I did not abuse the devices and I only used them scarcely. Neither were the devices left in the sun.
So, there must be something completely wrong with the plastic. Most probably too much weakener was used during the manufacturing process.

For this reason, I stopped developing on this platform. There are cheaper and better alternatives around.

Zontex, 2020-11-12:
Hello, which M5Stack device have you tested with? using M5Stack device with built-in battery or connected external battery shouldn't be an issue at all - could go and wakeup from deep sleep. hope it helps.

closed 2020-11-12 by Zontex

pedrocouto11, 2020-11-20:
I'm having the same issue. Is there a solution yet?

AbitTheGray, 2024-07-14:
Testing with two CoreInks. The one connected to USB wakes up OK but the one just on battery will never wake up from deep sleep. My wakeup interval are 10 minutes.