m5stack/M5Stack #35: SD Card Problems

ia2018, 2018-01-11:
Hello,
I recently received several (6 actually) M5Stack devices. After doing the capacitor mod, I've been able to download and test the supplied demos. Working with the FactoryTest sketch, I'm having problems mounting the SD Card. Here is my serial debug output:
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:137] sdCommand(): no token received
[W][sd_diskio.cpp:471] ff_sd_initialize(): GO_IDLE_STATE failed
[E][sd_diskio.cpp:739] sdcard_mount(): f_mount failed 0x(3)
Card Mount Failed
I've tried other SD Card sketches with no luck. I've tried formatting (Windows & SD Formatter App) three different 3 (64gb microSD) sd cards also no luck.
Any help would be appreciated!

Kabron287, 2018-02-25:
The same thing. Also LCD stops working.

abbey1501120, 2018-03-03:
I am also facing the same issue of 'Card Mount Failed' as below:

E (61) sdmmc_req: handle_idle_state_events unhandled: 00000004 00000000
E (63) sdmmc_cmd: sdmmc_card_init: send_op_cond (1) returned 0x107
Card Mount Failed

I am using M5stack (white color module) to test the SDMMC_Test example in Arduino. As I debug further I see that the code mentions the use of Pins 2, 4, 12, 13, 14, 15 for SD card, whereas as I see the schematics of M5stack , the SD card connector is placed on GPIO 4, 23, 18, 19 . I think this is the mismatch.
I have this confusion as where and in which file do I need to change the pin allocation  as I am not much into coding domain.
Thanks for reading and further help.

psatya111, 2018-04-02:
I have same problem.my sdcard pin configuration are  sck->18,miso->19,mosi->23,ss->5.please help me if any one solve this issue.

Dr4s1l, 2020-03-24:
Same issue here, seem's that the sdmmc exemple from idf is not working get my pin configuration on 18,23,19,4 and still not working. Getting issue that indicate some pull up resistors not connected

EeeeBin, 2020-03-26:
Some low-speed SD cards do not initialize properly
Please test another SD card (high speed)

mongonta0716, 2020-03-26:
They are connected via the TFCard (not SDCard) SPI interface used by the M5Stack.

The points to note are as follows

- There are some SD cards that are not supported.
- The format should be FAT32, not exFAT.

Also, it says in the manual that it's under 16GB.
If you use 64GB, you may be able to use it if you reformat it with FAT32 instead of exFAT.

Dr4s1l, 2020-03-27:
Wanted to use the m5 as a format SD tool, PIN configuration are not the right ones? So the sd card may have no FS on it. Using only the initial part of esp-idf sd_card exemple. Am i doing wrong?

Zontex, 2020-11-12:
Hello, could you please supply code example or more detailed information to reproduce the issue and see what's going on?

dme1sc, 2021-11-07:
Using M5Stack.h in M5 Stack ESP32 for Faces.
Here's my code in Loop:

  delay(30000);
  /*m5 lcd function*/
 M5.begin(); //Init M5Core. Initialize M5Core
  M5.Power.begin(); //Init Power module. Initialize the power module
                    /* Power chip connected to gpio21, gpio22, I2C device
                      Set battery charging voltage and current
                      If used battery, please call this function in your project */
  M5.Lcd.println("Hello World"; // Print text on the screen (string) Print text on the screen (string)

This is my response:
M5Stack initializing...[ 33747][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 33848][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 33949][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 34049][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[ 34050][W][sd_diskio.cpp:510] ff_sd_initialize(): GO_IDLE_STATE failed
[ 34051][E][sd_diskio.cpp:790] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[ 34060][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 34165][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 34266][W][sd_diskio.cpp:174] sdCommand(): no token received
[ 34366][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
OK

PierreRambaud, 2022-02-15:
I'm able to reproduce the issue too.
The SD card is properly found at first, but if I create anything on it like `SD.mkdir()`, the reset button makes the sd card unreadable.
```
[  3072][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x0d
[  3072][E][sd_diskio.cpp:614] ff_sd_status(): Check status failed
[  3376][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  3680][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
```

Tinyu-Zhao, 2024-07-23:
I used SanDisk's 16GTF card, the format is FAT32, and there was no problem in the test.
Please ensure 1: The TF card used is less than 16GB
2. The format is FAT32 format
[image]

closed 2024-07-23 by Tinyu-Zhao