m5stack/M5Stack #37: WiFiSetting example not working

KarlNorway, 2018-01-24:
I am experiencing that the example for WiFi setting, which I think is a good substitue for ESP8266 WiFi manager. Not working. 'I think that this might be to that fact that my SSID contains space. I tried to connect to my WiFi by just putting in ssid and passwd and that works. I'm thinking that is is something to do with the way that prefrences handles chars being stored.
` WiFi.begin(ssid, pwd);`

Zontex, 2020-11-12:
Could you please provide more information about what M5Stack device you are using and perhaps give an example of your code to see what's going on? docs.m5stack.com have detailed information hw to setup wifi on each device both in AP Mode which is pretty straight forward or manually in Arduino IDE or MicroPython, closing issue for now, if helped is still required feel free to open it.

closed 2020-11-12 by Zontex

KarlNorway, 2020-11-12:
@Zontex This was made some time ago. Let's see what I remember.. 

I had issues with space in the SSID. Going through the source I found that the space was not added as a character. After adding I got it working