m5stack/M5Stack #39: MPU9250 generate different values after every RESET, why?

Sterpa, 2018-01-27:
Dear All,
can anybody explane my, why MPU9250 generate different values after every RESET?
i use a basic example for MPU9250 with no changes, and my M5Stack with build-in MPU9250
Every turning RESET button i see a different values, like this:

```
ax = 0.92 ay = -154.54 az = 1003.05 mg
gx = -0.03 gy = 0.11 gz = -0.01 deg/s
mx = -9445 my = 2018 mz = 538 mG
q0 = 0.03 qx = -0.21 qy = 0.04 qz = 0.98
Yaw, Pitch, Roll: 169.37, 24.01, 4.62
rate = 0.13 Hz

RESET

ax = -1.71 ay = -0.18 az = 844.48 mg
gx = 0.04 gy = 0.08 gz = -0.02 deg/s
mx = -9438 my = 2007 mz = 527 mG
q0 = 0.03 qx = -0.05 qy = 0.05 qz = 1.00
Yaw, Pitch, Roll: 168.68, 6.41, 5.17
rate = 0.13 Hz

RESET

ax = -155.27 ay = -159.12 az = 997.80 mg
gx = -0.11 gy = 0.08 gz = -0.44 deg/s
mx = -9441 my = 2025 mz = 517 mG
q0 = 0.78 qx = -0.11 qy = 0.01 qz = 0.62
Yaw, Pitch, Roll: 67.79, 8.74, -8.84
rate = 900.86 Hz

RESET

ax = -1.77 ay = -157.35 az = 842.96 mg
gx = -0.02 gy = 0.04 gz = -0.01 deg/s
mx = -9425 my = 2013 mz = 491 mG
q0 = 0.77 qx = -0.06 qy = -0.06 qz = 0.64
Yaw, Pitch, Roll: 70.97, -0.74, -10.23
rate = 905.35 Hz
```

And so on...
The most strange is that Yaw, Pitch, Roll are different every reset, but the device is not moved!
Why accel and magnet are different every reset?
May be i should special calibrate accel and magnet?

0x1abin, 2018-01-30:
It is best to calibrate the magnetometer on startup, and the magnets in the bottom will interfere with it.

tuupola, 2018-02-11:
My guess is you are converting the the high and low byte as small endian.  6050 / 9250 is big endian.

dsprogis, 2018-02-12:
OMG, the infamous Intel Endian returns!  I thought we were rid of that!

AcOCrW, 2018-02-22:
Hi my friends.
can anybody explane me, why MPU9250 generate different values after every RESET?
i use a basic example for MPU9250 with no changes, and my M5Stack with built-in MPU9250.
 friends. my mpu9250 gyroscope sensor is constantly restarting M5 stackable .

AcOCrW, 2018-02-22:
Hi Sterpa ,
 did you solve this problem ?
Still I have problem with my M5 stack

quetzal, 2018-03-13:
Hello,

I've the same issue. Someone on m5stack forum pointed me this issue and this article to calibrate it: https://appelsiini.net/2018/calibrate-magnetometer/
I will see if it solve the problem.

izanette, 2018-04-10:
Any updates on this issue? Has anyone tried to port the https://github.com/kriswiner/MPU9250/blob/master/MPU9250_MS5637_AHRS_t3.ino to the M5Stack?

ifrew, 2018-05-11:
Hi Folks, finally figured it out. I have been using the mpu9250 on esp32 for a while now and just started using the m5stack.

The sample mpu9250.cpp driver in the stack is not resetting the mpu9250 device during startup. The modification is to the mpu9250.cpp file as follows. 

In the init routine change the value 0x00 to 0x80. This will reset the device to a set of know defaults at startup.

`void MPU9250::initMPU9250()
{
 // wake up device
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x00); // Clear sleep mode bit (6), enable all sensors 
  delay(100); // Wait for all registers to reset`

to 

`void MPU9250::initMPU9250()
{
 // wake up device
  writeByte(MPU9250_ADDRESS, PWR_MGMT_1, 0x80); // Clear sleep mode bit (6), enable all sensors 
  delay(100); // Wait for all registers to reset `

ifrew, 2018-05-12:
Hmm may have spoke too soon. The calibration routine resets the device, so an Inti followed by a calibration followed by an Init should work..but it is giving inconsistent values back.. hmm Referencing the issue on the developers repository #232

sukeshak, 2018-06-17:
[image]
I am having same issues. Testing now with processing script to figure out. 
Guess GPS or the magnet on base is causing interference.

Zontex, 2020-11-12:
Hello, tested the following code: https://github.com/m5stack/M5Stack/blob/master/examples/Basics/IMU/IMU.ino
Seems like there isn't any problems, it probably was a software issue that got fixed. make sure you have the latest M5Stack library version before testing it, regarding roll pitch and yaw, it takes time to stabilise so eventually after few seconds it will go back to the position as before the reset, it might require calibration. closing for now.

closed 2020-11-12 by Zontex