m5stack/M5Stack #40: M5 Speaker and WAV files

alanroche2015, 2018-01-28:
Hi all

Just wanted to see if anyone has played around with the internal speaker and playing WAV files ? Struggling to make anything that resembles audio working. 

Thanks

0x1abin, 2018-01-30:
Hi, thank you for your feedback, i will upload the audio player(wav and mp3) examples here.

alanroche2015, 2018-01-30:
Thanks that would be fantastic. Just spent another night playing around with no success. Looking forward to see where I’m going wrong.

0x1abin, 2018-01-30:
You can try this library:
https://github.com/earlephilhower/ESP8266Audio

alanroche2015, 2018-01-30:
I think I tried that library, all I got was something that kinda sounded like music with a load of static. I sounded like it was been amplified x 300.  Don’t suppose you’d have some sample .ino file you know works ?

alanroche2015, 2018-01-31:
I think i made some progress, i now have the M5 playing audio but it sounds like its playing it way to fast, here is my code so far... any ideas what it might be ??

`
#include <Arduino.h>
#ifdef ESP32
  #include <WiFi.h>
  #include "SPIFFS.h"
#else
  #include <ESP8266WiFi.h>
#endif

#include "AudioFileSourceID3.h"
#include "AudioGeneratorMP3.h"
#include "AudioOutputI2SNoDAC.h"
#include "AudioOutputI2S.h"
#include "AudioFileSourceSD.h"

#include "AudioGeneratorWAV.h"

#include <M5Stack.h>

AudioGeneratorWAV *mp3;
AudioFileSourceSD *file;
AudioOutputI2S *out;
AudioFileSourceID3 *id3;

void setup()
{
  M5.begin();

  M5.Speaker.setVolume(3);

  Serial.begin(115200);
  delay(1000);
  //SPIFFS.begin();
  Serial.printf("Sample MP3 playback begins...\n");
  file = new AudioFileSourceSD("/slow.wav");

  out = new AudioOutputI2S(0,true);
  

  mp3 = new AudioGeneratorWAV();
  mp3->begin(file,out);

}

void loop()
{
  M5.update();
  if (mp3->isRunning()) {
    M5.update();

    if (!mp3->loop()) {
      mp3->stop();
    }
  } else {
    Serial.printf("MP3 done\n");
    delay(1000);
  }
}
`

Zontex, 2020-11-12:
Hi Alan, is this issue still relevant? let me know if it is I will look into it. closing for now.

closed 2020-11-12 by Zontex

playground, 2021-02-07:
Hi, I'm trying to do something similar, how can I play the audio from for example: `https://translate.google.com/translate_tts?ie=UTF-8&q=Hello%2C+World%21&tl=en&client=tw-ob&ttsspeed=1` and store it for playback later?