m5stack/M5Stack #41: arduino-esp32 boards.txt assignment

ImpulseAdventure, 2018-02-05:
Hi --

I noticed that the name included in the **arduino-esp32** `boards.txt` section for **M5Stack** included hyphens in the `build.board=` assignment:
```
m5stack-core-esp32.build.board=M5Stack-Core-ESP32
```

When compiling under the Arduino IDE, the `build.board=` definition can usually be used as a macro target in a preprocessor conditional (eg. `#ifdef`). Unfortunately, hyphens are not valid characters in the macro name. As a result, attempting an `#ifdef M5Stack-Core-ESP32` will typically result in an `extra tokens at end of #ifdef directive` error/warning (and I believe the compiler will interpret the statement as `#ifdef M5Stack` instead).

This issue could potentially be avoided if the `boards.txt` submission were renamed without hyphens, eg:
```
m5stack-core-esp32.build.board=M5Stack_Core_ESP32
```
At the moment this may not cause many issues, but I suspect it could result in macro collisions if **M5Stack** were to release other variants (eg. `M5Stack-Core-zzz`) in the future. Anyways, just a minor suggestion.

0x1abin, 2018-02-06:
Thank you for your suggestion, this is very useful to me! I will correct it.

reaper7, 2018-05-11:
This issue should be closed
build.board is changed:
https://github.com/espressif/arduino-esp32/pull/1187
build.variant and corresponding variant folder also changed:
https://github.com/espressif/arduino-esp32/pull/1242

Zontex, 2020-11-12:
Issue closed.

closed 2020-11-12 by Zontex