m5stack/M5Stack #51: hi

ikapi, 2018-04-15:
Arduino:1.8.5 (Windows 7), 开发板："M5Stack-Core-ESP32, QIO, 80MHz, 921600, None"

In file included from C:\Users\Administrator\Documents\Arduino\libraries\M5Stack\examples\Modules\FACES\I2C_Keyboard\I2C_Keyboard.ino:1:0:

C:\Users\Administrator\Documents\Arduino\libraries\M5Stack\src/M5Stack.h:110:23: fatal error: esp_sleep.h: No such file or directory

compilation terminated.

exit status 1
为开发板 M5Stack-Core-ESP32 编译时出错。

so what shoudld i do?

tobozo, 2018-04-16:
if you're using Arduino as an ESP-IDF component [there's a fix for that](https://github.com/espressif/arduino-esp32/issues/633)

GitHub-Hugl, 2019-08-21:
Arduino:1.8.9 (Windows 10), 开发板："NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Enabled, 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

构建选项已变更，全部重新构建
C:\Users\Administrator\Documents\Arduino\libraries\M5Stack-master\src\M5Display.cpp:198:23: fatal error: rom/tjpgd.h: No such file or directory

 #include "rom/tjpgd.h"

                       ^

compilation terminated.

exit status 1
为开发板 NodeMCU 1.0 (ESP-12E Module) 编译时出错。

tobozo, 2019-08-21:
@GitHub-Hugl if you've selected `NodeMCU 1.0 (ESP-12E Module)` board instead of `M5Stack-Core-ESP32` this can happen :rofl:

Zontex, 2020-11-12:
You've probably selected the wrong board from the menu, make sure you select the right device corresponding to the one you own.

closed 2020-11-12 by Zontex