m5stack/M5Stack #53: Feature request: M5Stack library version tracking

tobozo, 2018-04-18:
Hello,

In order to prevent conflicts and path problems with pre-existing libraries, may I suggest M5Stack is bundled with a variable that contains its own version?

Currently, if I have a M5 project using qrcode.h, it works fine with 0.1.6 but breaks with 0.1.7.

This creates a situation where every M5 project using one of the new libs will break, or will have to maintain two codebases until the M5Stack 0.1.7 lands in the ESP32 environment.

In order to prevent such a SNAFU, M5 library should take care of the problem it created by giving the possibility to the developer or the compiler to avoid the breaking situation.

Suggested solution: "tattoo" all upcoming versions (including 0.1.7)

This could be done every time a new release is produced:

`echo "#define M5_LIB_VERSION F(\"$(git describe --tags --always --dirty)\")" > gitTagVersion.h`

and of course including it from somewhere in the stack:

`#include <gitTagVersion.h>`

so we can easily use something like this:

```
#ifdef M5_LIB_VERSION
  #include "utilities/qrcode.h"
#else 
  #include "qrcode.h
#endif 
```

and not being forced to maintain our project code every time M5 maintains its code :-)

0x1abin, 2018-04-19:
Hi,
Thank you so much for your suggestion, and sorry for the inconvenience. I added the gitTagVersion in next version :-)

tobozo, 2018-04-19:
working well, thanks for the quick implementation 👍

closed 2018-04-19 by tobozo