m5stack/M5Stack #55: Compilation errors in FactoryTest example (min/max macros and WiFi.h)

wtakuo, 2018-05-03:
The compilation of `FactoryTest` example fails with the latest `M5Stack` and `arduino-esp32` libraries due to the missing declaration of `min` used in the source code.  Actually, `min` and `max` are defined as macros in `Arduino.h` but are undefined in `WiFiClient.h` included indirectly from `WiFi.h`.  So if we include `WiFi.h` after `Arduino.h` (as in the source code of `FactoryTest`), we may encounter missing-definition errors for these macros. 

A possible fix is to use `_min` instead of `min`.  In `Arduino.h`, macros `_max` and `_min` are defined just for cases that `min` and `max` are unusable.  Another solution is to define `min` macro again in the source code.

tobozo, 2018-05-03:
Display.h also gives problems with min/max macros as seen in this [travis log](https://travis-ci.org/tobozo/M5Stack-SD-Updater/builds/372781647#L4044)

wtakuo, 2018-05-06:
This issue itself was resolved in [the last commit ](https://github.com/m5stack/M5Stack/commit/728ffce4f6d495c3ce18e0a1a7d492353e3eedeb) in which PR #57 was merged.
So I think I should close this issue.

The [tobozo](https://github.com/tobozo)'s comment illuminates another important issue. It seems that the following (re-)definitions of min/max macros in Display.h conflict with the names of member functions of the standard vector class. 

```C++
#ifndef max
#define max(a, b) (((a) > (b)) ? (a) : (b))
#endif
#ifndef min
#define min(a, b) (((a) < (b)) ? (a) : (b))
#endif
```

I think these unnecessary macro definitions in Display.h should be removed.

closed 2018-05-06 by wtakuo