m5stack/M5Stack #68: Lcd.drawBitmap color in ver. 2

vvs551, 2018-05-22:
I use Lcd.drawBitmap function to draw icons with different colors but the same teplate on the screen:

> M5.Lcd.drawBitmap(int16_t x, int16_t y, const uint8_t bitmap[], int16_t w, int16_t h, uint16_t color); Draw bitmap with points X0 and Y0, the bit array of brightness from 0 to 255, width W, height H and color from 0 to 65535

But there is absent option "color" in ver.2
How to change bitmap color with drawBitmap function?

tobozo, 2018-05-22:
Looks like it's gone after the last upgrade to TFT_eSPI.
Here's the [original code](https://github.com/m5stack/M5Stack/blob/aab5cf45bd1ab4fffd9e683ad47d19ceb3e0ac47/src/utility/Display.cpp#L927) as you can see it's only a wrapper around drawPixel() and can be easily re-created.

I can't submit a PR for that as I'm unsure where this should fit since the file is now gone.

@0x1abin @wtakuo any idea where this should be done?

0x1abin, 2018-05-23:
Maybe can use the TFT_eSPI drawXBitmap function to replace the old drawBitmap.
drawXBitmap(int16_t x, int16_t y, const uint8_t *bitmap, int16_t w, int16_t h, uint16_t color);
drawXBitmap(int16_t x, int16_t y, const uint8_t *bitmap, int16_t w, int16_t h, uint16_t fgcolor, uint16_t bgcolor);
https://github.com/m5stack/M5Stack/blob/master/src/utility/In_eSPI.h#L543

vvs551, 2018-05-28:
There is an error in showing of the icon with drawXBitmap function
Here is a simple example of the icon with percent (%) symbol, but the result on the screen is absolutely incorrect:

const uint16_t lbl1Width = 30;
const uint16_t lbl1Height = 30;
uint8_t lbl1[120] PROGMEM= {
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x06, 0x00, 0x60, 
    0x00, 0x1f, 0x80, 0xc0, 
    0x00, 0x19, 0x81, 0xc0, 
    0x00, 0x30, 0xc1, 0x80, 
    0x00, 0x30, 0xc3, 0x00, 
    0x00, 0x30, 0xc6, 0x00, 
    0x00, 0x30, 0xc6, 0x00, 
    0x00, 0x19, 0x8c, 0x00, 
    0x00, 0x1f, 0x9c, 0x00, 
    0x00, 0x0e, 0x38, 0x40, 
    0x00, 0x00, 0x31, 0xf0, 
    0x00, 0x00, 0x63, 0x30, 
    0x00, 0x00, 0xc6, 0x18, 
    0x00, 0x00, 0xc6, 0x18, 
    0x00, 0x01, 0x86, 0x18, 
    0x00, 0x03, 0x06, 0x18, 
    0x00, 0x03, 0x06, 0x30, 
    0x00, 0x06, 0x03, 0xb0, 
    0x00, 0x0c, 0x03, 0xe0, 
    0x00, 0x04, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00, 
    0x00, 0x00, 0x00, 0x00
};

tobozo, 2018-05-28:
how does it work if you just use a regular function ?

```
void drawBitmap(int16_t x, int16_t y, const uint8_t *bitmap, int16_t w, int16_t h, uint16_t color) {
  int32_t i, j, byteWidth = (w + 7) / 8;
  for (j = 0; j < h; j++) {
    for (i = 0; i < w; i++) {
      if (pgm_read_byte(bitmap + j * byteWidth + i / 8) & (128 >> (i & 7))) {
        M5.Lcd.drawPixel(x + i, y + j, color);
      }
    }
  }
}
```

vvs551, 2018-05-28:
What do you mean? Is every point in monochrome image need a byte? )
There is an icon with % sign:
[image]
Here is a conversion:
[image]
[image]
And there was not any problem with drawBitmap in previous version
How to correct situation?

tobozo, 2018-05-28:
>  And there was not any problem with drawBitmap in previous version

Yep, and I pasted the missing drawBitmap function you're complaining for, does this function work for you?

If it does work, then you've corrected the situation.

Also if that snippet from your "simple example code" is correct:

    uint8_t lbl1[120]

then those values are just wrong: 

    const uint16_t lbl1Width = 30;
    const uint16_t lbl1Height = 30;

and should be respectively 10x12 or 8x15

vvs551, 2018-05-28:
ThankU! Function is working good!
I'm agree that 30*30 = 900;  900/8 = 112.5
That's why the length 113 will be enough (not 120)
But converter give us 120 ))

tobozo, 2018-05-28:
yup I counted on my fingers and realized how wrong I was and I deleted that part of my comments :-)

so if the function is still working even after migrating to TFT_eSPI, there's still hope it can be re-integrated \o/ (poke @0x1abin )

However, the function name `drawBitmap` sounds a bit misleading, maybe `drawMonochrome` would be a better fit?

Zontex, 2020-11-12:
Sounds like problem got sorted out, I'm closing this issue for now.

closed 2020-11-12 by Zontex