m5stack/M5Stack #71: How to decide which analog port to use (male/female)

chetanpt, 2018-06-06:
Hi 

I am using pin 35 to read fsr sensor data. FSR sensor is connected to female port. 
I do not receive any data on female port whereas on the male port . 

How to we decide which ports to use and how to change that.

I am using arduino ide. 

Thanks

chetanpt, 2018-06-07:
@tobozo 
Thanks for the information 

In my case female ADC pin does not get data whereas I receive data on male ADC pins. 

So my question is how do I resolve this, so that I can receive data on both pins without worrying  ?

Thanks
-Chetan

tobozo, 2018-06-07:
Could it be broken ?

[image]

chetanpt, 2018-06-08:
Hi 

In my case every alternate day the M5 stack does not receive data. It always shows  0.

I use following code on platformio

#include <M5Stack.h>

int fsrpin = 35;
int wristforce;

void setup() {
    // put your setup code here, to run once:

    M5.begin();
    pinMode(fsrpin, INPUT);
}

void loop() {
    // put your main code here, to run repeatedly:
    M5.Lcd.clearDisplay();
    M5.Lcd.setTextColor(RED);
    M5.Lcd.setTextSize(3);
    M5.Lcd.setCursor(15, 15);

    wristforce = analogRead(fsrpin);

    Serial.println(wristforce);
    M5.Lcd.print(wristforce);

    }

    M5.update();
    delay(500);
}

chetanpt, 2018-06-08:
Hi @tobozo 

I found the root cause after troubleshooting and searching on google

I connect my FSR to 5V but the 5V pin is not predictable sometimes it give ~ 4.5v and sometimes it does not. I do not know the reason. 

But the 3.3 v and +ve terminal of grove connector (~4.5v) are more reliable, I think.
I reffered following discussion on forum
http://forum.m5stack.com/topic/153/connecting-m5stack-to-an-mq-2-smoke-gas-sensor/2

Thanks for your help