m5stack/M5Stack #73: m5stack_startup_music not defined in Speaker example

jadr2ddude, 2018-07-02:
It appears that the startup music is in the FactoryTest example and is therefore inaccessible to the Speaker example. This music must be copied to the Speaker example to get this to work.

jadr2ddude, 2018-07-02:
Appears that issue #69 is similar

0x1abin, 2018-07-03:
Thanks for your feedback, is an old way to play music, I remove it, but you can find out this file at [this](https://github.com/m5stack/M5Stack/blob/master/examples/Basics/FactoryTest/startup_music.cl)
If you want play music, reference: https://github.com/m5stack/M5Stack/tree/master/examples/Advanced/Audio

closed 2018-07-03 by jadr2ddude

tobozo, 2019-05-18:
[permalink to 'this'](https://github.com/m5stack/M5Stack/blob/7e863fb9934bcb25f15afa720e9299b8a8e9a8fd/examples/Basics/FactoryTest/startup_music.c)