m5stack/M5Stack #74: Battery API

vshymanskyy, 2018-07-06:
Is there any way to get the battery charge level/status?

0x1abin, 2018-07-07:
Sorry, the hardware nonsupport get the battery charge level.

vshymanskyy, 2018-07-07:
@0x1abin I'm talking about FIRE, which presumably has `IP5306_I2C`

0x1abin, 2018-07-09:
The FIRE can get the battery charge status by i2c, but can't get level.

0x1abin, 2018-07-10:
The power IC IP5306 datasheet is Chinese :D ..., I write a simple function: 
https://github.com/m5stack/M5Stack/blob/m5stack-fire/src/M5Stack.cpp#L70

plex555, 2018-12-19:
IP5306 registers map:
https://yadi.sk/i/dRp0HTF4oPCo2g

IP5209 IP5109 IP5207 IP5108 registers map:
https://yadi.sk/i/oNhqTei36jFwSg

A bit more about IP5108, but only russian language (google translater help you):
https://wiki.iarduino.ru/page/Battery_Shield/

Battery Shield Libary for arduino (with Register description):
http://iarduino.ru/file/344.html

P.S. If somebody understand power IC (such as IP5*) programming, can you help save my settings in default register. If I put power off, all my I2C settings will set default and don't want each time connect my power chip to arduino and change settings.

lovyan03, 2019-03-11:
I found it.
If manufacturing time is newer than FIRE release, it is also possible for GRAY and BASIC.

https://mobile.twitter.com/lovyan03/status/1104548868199337984

[image]

lovyan03, 2019-03-11:
int8_t getBatteryLevel()
{
  Wire.beginTransmission(0x75);
  Wire.write(0x78);
  if (Wire.endTransmission(false) == 0
   && Wire.requestFrom(0x75, 1)) {
    switch (Wire.read() & 0xF0) {
    case 0xE0: return 25;
    case 0xC0: return 50;
    case 0x80: return 75;
    case 0x00: return 100;
    default: return 0;
    }
  }
  return -1;
}

lovyan03, 2019-03-11:
`int8_t getBatteryLevel()
{
  Wire.beginTransmission(0x75);
  Wire.write(0x78);
  if (Wire.endTransmission(false) == 0
   && Wire.requestFrom(0x75, 1)) {
    switch (Wire.read() & 0xF0) {
    case 0xE0: return 25;
    case 0xC0: return 50;
    case 0x80: return 75;
    case 0x00: return 100;
    default: return 0;
    }
  }
  return -1;
}
`

lovyan03, 2019-03-11:
`
int8_t getBatteryLevel()  
{  
  Wire.beginTransmission(0x75);  
  Wire.write(0x78);  
  if (Wire.endTransmission(false) == 0  
   && Wire.requestFrom(0x75, 1)) {  
    switch (Wire.read() & 0xF0) {  
    case 0xE0: return 25;  
    case 0xC0: return 50;  
    case 0x80: return 75;  
    case 0x00: return 100;  
    default: return 0;  
    }  
  }  
  return -1;  
}  
`

lovyan03, 2019-03-11:
```
int8_t getBatteryLevel()
{
  Wire.beginTransmission(0x75);
  Wire.write(0x78);
  if (Wire.endTransmission(false) == 0
   && Wire.requestFrom(0x75, 1)) {
    switch (Wire.read() & 0xF0) {
    case 0xE0: return 25;
    case 0xC0: return 50;
    case 0x80: return 75;
    case 0x00: return 100;
    default: return 0;
    }
  }
  return -1;
}
```

sakabin, 2019-03-11:
It looks great, I will check it and add put it to power.c

KOsterOk2, 2019-03-11:
BASIC no work. Return -1

lovyan03, 2019-03-11:
[image]

It will not work if the manufacturing date is old.
If it is new, BASIC and GRAY will also work.

KOsterOk2, 2019-03-11:
My BASIC is 2018.03 . Can you post all Code here ?

lovyan03, 2019-03-11:
Perhaps the differences between versions of various libraries may affect you.
If there is a 0x75 response with "I2C Scanner" it should work.

I cut out only the necessary parts from this project.
https://github.com/lovyan03/M5Stack_LovyanLauncher

lovyan03, 2019-03-11:
Did you run "Wire.begin()" ?
```
void setup() {
  M5.begin();
  Wire.begin();
・・・
```

KOsterOk2, 2019-03-11:
Now it works! Thanks you!
```
#include <M5Stack.h>

int8_t getBatteryLevel()
{
  Wire.beginTransmission(0x75);
  Wire.write(0x78);
  if (Wire.endTransmission(false) == 0
   && Wire.requestFrom(0x75, 1)) {
    switch (Wire.read() & 0xF0) {
    case 0xE0: return 25;
    case 0xC0: return 50;
    case 0x80: return 75;
    case 0x00: return 100;
    default: return 0;
    }
  }
  return -1;
}

void setup() {
    M5.begin();
    Wire.begin();
  }

void loop() {
  delay(1000);
  M5.Lcd.clear();
  M5.Lcd.setTextColor(GREEN);
  M5.Lcd.setCursor(0, 20);
  M5.Lcd.setTextSize(2);
  M5.Lcd.print("Battery Level: ");
  M5.Lcd.print(getBatteryLevel());
  M5.Lcd.print("%");
}

```

closed 2019-03-11 by vshymanskyy

oteroosc, 2019-05-04:
Will it work with the 700mAh battery board attached?

tatsuya-takahashi, 2020-01-24:
> Will it work with the 700mAh battery board attached?

It worked for me with 700mAh battery.

tialm, 2020-05-18:
Would this work in the 2018.2A version?

guido-visser, 2020-08-04:
Is there any way to make this work on the M5Stick-C as well?

Neil-McK, 2021-02-17:
Update: Try M5.Power.getBatteryLevel() and M5.Power.isChargeFull() functions in standard M5Stack.h library.

Knockoi, 2023-09-27:
Excuse me. Can I help you? Do you have a more detailed Arduino program? I don't know much about the I2C mechanism of the IP series charging ICs, I just need to display the battery level.