m5stack/M5Stack #82: No display, just a pulsing static noise

watahboy, 2018-07-27:
I ordered the m5stack as soon as it was available to me and had a sketch showing me BTC prices from an exchange and it worked for about 3 months. Now the m5stack is just making a static/ scratching noise about every half second and otherwise is showing no signs of life. I tried reflashing the sketch and I am getting the same result. Did my unit fry?

joeybab3, 2019-01-10:
Probably if you're no longer able to upload sketches to it. Try reaching out to the site where you bought it from.

sulolullero, 2019-09-21:
Hi @watahboy! Did you resolve the issue? I'm having same problems currently. Cheers, Mikko

cmoncada, 2019-11-07:
I have same problem! any fix or clue!!!

EeeeBin, 2019-11-07:
For some reasons, our BTC price acquisition service has been closed, try find other APIs to get BTC prices

cmoncada, 2019-11-07:
I Only configured wifi, restarted M5stack, but now is only making noise and screen is off.

EeeeBin, 2019-11-07:
Could you show the USB UART debug info

flouroantimonic-acid, 2019-12-26:
I got this same problem, and I just got mine. I can upload and serial works fine. No idea why. I'm kinda pissed.

flouroantimonic-acid, 2019-12-26:
It seems that the speaker works fine (tested with the Speaker example), but the Display example does nothing.

Gitshaoxiang, 2019-12-26:
any product questions.  pls contact the our store.  like AliExpress

closed 2020-11-12 by Zontex