m5stack/M5Stack #85: package_m5stack_index.json file signature verification failed

sukeshak, 2018-08-16:
Getting this error always and unable to configure or download the package automatically

http://www.m5stack.com/download/package_m5stack_index.json file signature verification failed. File ignored.

corq, 2018-08-17:
I too have received this error recently and was unable to resolve.

plugboy, 2018-10-18:
Just bought the gray development kit and got the same issue.
Will there be a fix soon?

ranjanjan, 2018-10-20:
same as plug

plugboy, 2018-10-20:
Eventually get the device to work with Arduino IDE with this guide https://github.com/m5stack/m5stack-board-support/blob/master/doc/windows.md

And I also install the driver here https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers

quantumbeat, 2019-03-04:
the same today - worked some other day on some other computer ..

sakabin, 2019-03-04:
Hello, try to test https://docs.m5stack.com/#/en/quick_start/m5core/m5stack_core_get_started_Arduino_Windows

quantumbeat, 2019-03-04:
It works! Is listed when installing ESP32. Thanks for quick

closed 2019-03-04 by sakabin