m5stack/M5Stack #86: beep volume

Keoni80, 2018-08-16:
the default volume for beep is too loud,, tried with 
M5.Speaker.setVolume(1);
M5.Speaker.update();
but still in same volume level,, how can edit the beep volume?

robbygallo, 2018-08-30:
I have the same problem. 
Please provide to solve it, my collegues hate me...

robbygallo, 2018-08-30:
I have found the solution in a chess game
setVolume only work with playMusic

`
uint8_t music[1000]; 
void tone_volume(uint16_t frequency, uint32_t duration) { 
  float interval=0.001257*float(frequency);  
  float phase=0;
  for (int i=0;i<1000;i++) {
   music[i]=127+126*sin(phase);  
   phase+=interval;
  }   
  music[999]=0;
  int remains=duration;
  for (int i=0;i<duration;i+=200) {   
   if (remains<200) {
    music[remains*999/200]=0;
   }
   M5.Speaker.playMusic(music,5000);  
   remains-=200;
  } 
}
//****************************
void beep(int leng) {
  M5.Speaker.setVolume(1);
  tone_volume(1000, leng); 
}  

beep(1000); // 1sec
`

gadjet, 2018-08-30:
robbygallo,
Thanks for this information but this code doesn't compile and I'm not expert enough to understand why there are issues in lines 3, 6 and 13, I'm guessing there's an operator missing ??

Any help would be appreciated.
Thanks

robbygallo, 2018-08-31:
I don't know your detail error.
This code work for me

> #include <M5Stack.h>
> 
> bool soundOn = true;
> 
> //////////////////////////////////////////////////////////////////////////////
> ///////////////////////////////////  BEEP  ///////////////////////////////////
> //////////////////////////////////////////////////////////////////////////////
> uint8_t music[1000]; 
> void tone_volume(uint16_t frequency, uint32_t duration) { 
>   float interval=0.001257*float(frequency);  
>   float phase=0;
>   for (int i=0;i<1000;i++) {
>    music[i]=127+126*sin(phase);  
>    phase+=interval;
>   }   
>   music[999]=0;
>   int remains=duration;
>   for (int i=0;i<duration;i+=200) {   
>    if (remains<200) {
>     music[remains*999/200]=0;
>    }
>    M5.Speaker.playMusic(music,5000);  
>    remains-=200;
>   } 
> }
> void beep(int leng) {
>   if (soundOn) M5.Speaker.setVolume(1); else M5.Speaker.setVolume(0);            
>   tone_volume(1000, leng); 
> }  
> 
> 
> ////////////////////////////////////////////////////////////////////////////////
> ///////////////////////////////////  SETUP  ////////////////////////////////////
> ////////////////////////////////////////////////////////////////////////////////
> void setup(void) {
>   M5.begin();
> 
>   beep(100);
> }
> 
> 
> ////////////////////////////////////////////////////////////////////////////////
> ///////////////////////////////////  LOOP  /////////////////////////////////////
> ////////////////////////////////////////////////////////////////////////////////
> void loop() {
> 
> }
>

gadjet, 2018-08-31:
Hi,
The Arduino compiler has problems with: -
`float interval=0.001257float(frequency);` the ..... 0.001257float... the 7 next to the f
`music[i]=127+126sin(phase);` the ..... 126sin(... the 6 next to the sin
`music[remains*999/200]=0;` is now OK because there is now a * between the "remains and 999" which wasn't in the original code.

Should all the places where there is no gap etc have an * ?

gadjet, 2018-08-31:
I added a * to all the items that failed the compile and it now compiles and works OK, thanks for sharing.
```
void tone_volume(uint16_t frequency, uint32_t duration) {
float interval=0.001257 * float(frequency);
float phase=0;
for (int i=0;i<1000;i++) {
music[i]=127+126 * sin(phase);
phase+=interval;
}
music[999]=0;
int remains=duration;
for (int i=0;i<duration;i+=200) {
if (remains<200) {
music[remains * 999/200]=0;
}
M5.Speaker.playMusic(music,5000);
remains-=200;
}
}
```

spawnrider, 2019-01-30:
Same issue on my **M5Stack** (Basic black edition) with the following code :
```
M5.Speaker.setVolume(1);
M5.Speaker.update();
M5.Speaker.beep();
```

sboger, 2019-04-05:
You need to issue a begin() first...

  M5.Speaker.begin();
  M5.Speaker.setVolume(10);
  M5.Speaker.update();
  M5.Speaker.beep();

cristian-spiescu, 2020-10-03:
Hi guys. Hi M5Stack. A fix for this issue wouldn't be too bad! Any plans to provide such?

Zontex, 2020-11-12:
Hello, I've tested the following code with M5Stack-Core-ESP32 and it works, can set the volume:
`#include <M5Stack.h>

#define NOTE_D0 -1
#define NOTE_D1 294
#define NOTE_D2 330
#define NOTE_D3 350
#define NOTE_D4 393
#define NOTE_D5 441
#define NOTE_D6 495
#define NOTE_D7 556

#define NOTE_DL1 147
#define NOTE_DL2 165
#define NOTE_DL3 175
#define NOTE_DL4 196
#define NOTE_DL5 221
#define NOTE_DL6 248
#define NOTE_DL7 278

#define NOTE_DH1 589
#define NOTE_DH2 661
#define NOTE_DH3 700
#define NOTE_DH4 786
#define NOTE_DH5 882
#define NOTE_DH6 990
#define NOTE_DH7 112

void setup() {
  // Initialize the M5Stack object
  M5.begin();
  
  /*
    Power chip connected to gpio21, gpio22, I2C device
    Set battery charging voltage and current
    If used battery, please call this function in your project
  */
  M5.Power.begin();

  // set volume for the speaker
  M5.Speaker.setVolume(1);
  M5.Speaker.update();

  M5.Lcd.printf("M5Stack Speaker test:\r\n");
}

void loop() {
  if(M5.BtnA.wasPressed()) {
    M5.Lcd.printf("A wasPressed \r\n");
    M5.Speaker.beep(); //beep
  }

  if(M5.BtnB.wasPressed())
  {
    M5.Lcd.printf("B wasPressed \r\n");
    M5.Speaker.tone(NOTE_DH2, 200); //frequency 3000, with a duration of 200ms
  }

  M5.update();
}`
Make sure to update the Arduino m5stack library to the last version before you try this code.

closed 2020-11-12 by Zontex