m5stack/M5Stack #90: ESPcam demo compilation error

ModMike, 2018-09-05:
When I try to compile the ESP32 cam demo I get the following error:

```
CC /Users/Michael/Esp/esp32-cam-demo/build/camera/xclk.o
/Users/Michael/Esp/esp32-cam-demo/components/camera/./xclk.c: In function 'camera_enable_out_clock':
/Users/Michael/Esp/esp32-cam-demo/components/camera/./xclk.c:31:12: error: 'ledc_channel_config_t {aka struct <anonymous>}' has no member named 'hpoint'
     ch_conf.hpoint = 0;
            ^
make[1]: *** [xclk.o] Error 1
make: *** [component-camera-build] Error 2
```

charlie-x, 2018-09-09:
try branch >3.1 of esp-idf

ModMike, 2018-09-09:
I am trying to git the branch but it does not work:

git clone --recursive /tree/release/v3.1  https://github.com/espressif/esp-idf.git

Hoe do I do it?  If I don't use Git it tells me it's not a git repository when I try to make

ModMike, 2018-09-09:
I think I figured it out.  I cloned using GitHub desktop then switched branches in GitHub desktop.  Compiled with no errors.  Is that the correct procedure?

charlie-x, 2018-09-09:
that'll work.

Zontex, 2020-11-12:
Seems like figured it out, I'm closing it for now.

closed 2020-11-12 by Zontex