m5stack/M5Stack #91: M5Stack Fire crashes with PSRAM enabled

Akaprog, 2018-09-11:
1) Crashes doesn't occur for every code, but for specific cases. Disabling PSRAM in Arduino or changing code helps.

2) Device undergoing them passes M5Fire-psram-tests (so we shall assume memory is OK)

3) I've created few demos which could be used to reproduce issue (at least it happens with both M5Fire kits that I have and 2 kits my friend have ordered after me)

4) Not reproducable on any other ESP32 board that I have

Examples below:
https://github.com/Akaprog/M5Fire-Experiments/blob/master/M5Fire_fail/M5Fire_fail.ino

https://github.com/Akaprog/M5Fire-Experiments/blob/master/M5fire_fail2/M5fire_fail2.ino

sakabin, 2018-09-12:
you can use
`M5.begin(true, false, false);`  instead `M5.begin()；`
it will be ok, because in M5.begin() is enable serial already

Akaprog, 2018-09-14:
@sakabin Yes, thanks this will look better, I just have started porting my code to M5stack from some older examples :(

Anyway, crashes are still reproducable. They seem almost random to me - some of mine older esp32 projects do crash or need some small adjustments (or disabling PSRAM) to work stable with M5Fire.

Zontex, 2020-11-12:
Is this issue still relevant? seems like the new library should get it fixed, make sure you have the latest code and library installed. closing for now, if still relevant feel free to open the case and I will look into it.

closed 2020-11-12 by Zontex