m5stack/M5Stack #100: Display faulty? - how can i fix it...?

timtimsson, 2018-10-31:
Hi,
I got my m5stack yesterdy with a white screen.
To test the unit I burn the orginal firmware with multiple usb cable - only a white screen :(
My "old" M5Stackgrey working perfect with the same binaries?!
If i use a testfile with different brigness you can see color lines?
How can i fix this? Or can I get some help to confirm my suspicion?
Thanks!

https://youtu.be/f2JZdLe4wzA

sakabin, 2018-10-31:
hello, 
try to test  arduino **examples->Basics->FactoryTest**

timtimsson, 2018-10-31:
Hello,
thanks for your reaktion!
But also nothing to see - only ghosts....
https://youtu.be/sw7ARHekSlg

sakabin, 2018-10-31:
You can contact m5 after-sales service,  replace it

timtimsson, 2018-10-31:
Do you have a contact or direktly over aliexpress?
Thanks a lot...

sakabin, 2018-10-31:
direktly over aliexpress