m5stack/M5Stack #109: esp32-cam-demo i2c write error with Platformio

dontsovcmc, 2018-11-27:
I compile and flash your example and see i2c errors in the output.
I'm a newbei in ESP32, so maybe I do something wrong?

c_cpp_properties.json file: https://pastebin.com/aqXTmRUV

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5696
ho 0 tail 12 room 4
load:0x40078000,len:7768
ho 0 tail 12 room 4
load:0x40080000,len:5892
entry 0x40080318
␛[0;32mI (32) boot: ESP-IDF 3.30101.0 2nd stage bootloader␛[0m
␛[0;32mI (32) boot: compile time 22:22:57␛[0m
␛[0;32mI (32) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (36) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (41) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (45) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (49) boot: Partition Table:␛[0m
␛[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (74) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (82) boot: End of partition table␛[0m
␛[0;32mI (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x15770 ( 87920) map␛[0m
␛[0;32mI (126) esp_image: segment 1: paddr=0x00025798 vaddr=0x3ffc0000 size=0x03278 ( 12920) load␛[0m
␛[0;32mI (131) esp_image: segment 2: paddr=0x00028a18 vaddr=0x3ffc3278 size=0x00000 (     0) load␛[0m
␛[0;32mI (133) esp_image: segment 3: paddr=0x00028a20 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (142) esp_image: segment 4: paddr=0x00028e28 vaddr=0x40080400 size=0x071e8 ( 29160) load␛[0m
␛[0;32mI (162) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x67a8c (424588) map␛[0m
␛[0;32mI (313) esp_image: segment 6: paddr=0x00097aac vaddr=0x400875e8 size=0x09dcc ( 40396) load␛[0m
␛[0;32mI (329) esp_image: segment 7: paddr=0x000a1880 vaddr=0x400c0000 size=0x00000 (     0) load␛[0m
␛[0;32mI (329) esp_image: segment 8: paddr=0x000a1888 vaddr=0x50000000 size=0x00000 (     0) load␛[0m
␛[0;32mI (346) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (346) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (348) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (351) cpu_start: Starting app cpu, entry point is 0x40081e28␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (362) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (369) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM␛[0m
␛[0;32mI (375) heap_init: At 3FFC92C0 len 00016D40 (91 KiB): DRAM␛[0m
␛[0;32mI (381) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (387) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (393) heap_init: At 400913B4 len 0000EC4C (59 KiB): IRAM␛[0m
␛[0;32mI (400) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (82) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;32mI (145) gpio: GPIO[15]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 ␛[0m
␛[0;31mE (685) sccb: i2c write reg=0x4a err=0x107␛[0m
␛[0;31mE (1685) sccb: i2c write reg=0x21 err=0x107␛[0m
␛[0;31mE (1685) sccb: i2c write reg=0x24 err=0xffffffff␛[0m
␛[0;31mE (2685) sccb: i2c write reg=0x25 err=0x107␛[0m
␛[0;31mE (2685) sccb: i2c write reg=0x26 err=0xffffffff␛[0m
␛[0;31mE (2685) sccb: i2c write reg=0x48 err=0xffffffff␛[0m
␛[0;31mE (3685) sccb: i2c write reg=0x49 err=0x107␛[0m
␛[0;31mE (3685) sccb: i2c write reg=0x5c err=0x107␛[0m
␛[0;31mE (3685) sccb: i2c write reg=0x63 err=0xffffffff␛[0m
␛[0;31mE (3685) sccb: i2c write reg=0x46 err=0xffffffff␛[0m
␛[0;31mE (3685) sccb: i2c write reg=0x47 err=0xffffffff␛[0m
␛[0;31mE (3695) sccb: i2c write reg=0x0c err=0xffffffff␛[0m
␛[0;31mE (3705) sccb: i2c write reg=0x5d err=0x107␛[0m
␛[0;31mE (3705) sccb: i2c write reg=0x5e err=0x107␛[0m
␛[0;31mE (3705) sccb: i2c write reg=0x5f err=0x107␛[0m
␛[0;31mE (3715) sccb: i2c write reg=0x60 err=0x107␛[0m
␛[0;31mE (3715) sccb: i2c write reg=0x61 err=0xffffffff␛[0m
[... 229 lines cut ...]
    -DCONFIG_SPIRAM_CACHE_WORKAROUND
    -DCORE_DEBUG_LEVEL=4
    -DCONFIG_OV2640_SUPPORT
    -DCONFIG_OV7725_SUPPORT
    -DCONFIG_ESP_WIFI_MODE_AP=1
    -DCONFIG_ESP_WIFI_SSID="\"M5Cam\""
    -DCONFIG_ESP_WIFI_PASSWORD="\"\""
    -DCONFIG_MAX_STA_CONN=1
    -DCONFIG_XCLK_FREQ=20000000
    -DCONFIG_D0=17
    -DCONFIG_D1=35
    -DCONFIG_D2=34
    -DCONFIG_D3=5
    -DCONFIG_D4=39
    -DCONFIG_D5=18
    -DCONFIG_D6=36
    -DCONFIG_D7=19
    -DCONFIG_XCLK=27 
    -DCONFIG_PCLK=21
    -DCONFIG_VSYNC=22
    -DCONFIG_HREF=26
    -DCONFIG_SDA=25
    -DCONFIG_SCL=23
    -DCONFIG_RESET=15
```

dontsovcmc, 2018-11-27:
Very interesting: I compiled esp-com-demo with ESP-IDF (without Platformio).
It works for about 3 resets. But after 4th reset I see same i2c write error:

```
␛[0;32mI (87202) camera: Frame 452 done in 156 ms␛[0m
␛[0;32mI (87362) camera: Frame 453 done in 156 ms␛[0m
␛[0;32mI (87522) camera: Frame 454 done in 156 ms␛[0m
␛[0;32mI (87682) camera: Frame 455 done in 156 ms␛[0m
␛[0;32mI (87842) camera: Frame 456 done in 156 ms␛[0m
␛[0;32mI (88002) camera: Frame 457 done in 156 ms␛[0m
␛[0;32mI (88162) camera: Frame 458 done in 157 ms␛[0m
␛[0;32mI (88322) camera: Frame 459 done in 156 ms␛[0m
␛[0;32mI (88482) camera: Frame 460 done in 156 ms␛[0m
␛[0;33mW (88482) http_server: Expected response size: 4480, actual: 0␛[0m
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5824
ho 0 tail 12 room 4
load:0x40078000,len:9176
ho 0 tail 12 room 4
load:0x40080000,len:6008
entry 0x4008032c
␛[0;32mI (31) boot: ESP-IDF v3.1.1-dirty 2nd stage bootloader␛[0m
␛[0;32mI (31) boot: compile time 01:35:05␛[0m
␛[0;32mI (32) boot: Enabling RNG early entropy source...␛[0m
␛[0;32mI (36) boot: SPI Speed      : 40MHz␛[0m
␛[0;32mI (41) boot: SPI Mode       : DIO␛[0m
␛[0;32mI (45) boot: SPI Flash Size : 4MB␛[0m
␛[0;32mI (49) boot: Partition Table:␛[0m
␛[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length␛[0m
␛[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000␛[0m
␛[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000␛[0m
␛[0;32mI (74) boot:  2 factory          factory app      00 00 00010000 00100000␛[0m
␛[0;32mI (82) boot: End of partition table␛[0m
␛[0;32mI (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x169f8 ( 92664) map␛[0m
␛[0;32mI (128) esp_image: segment 1: paddr=0x00026a20 vaddr=0x3ffb0000 size=0x034e4 ( 13540) load␛[0m
␛[0;32mI (133) esp_image: segment 2: paddr=0x00029f0c vaddr=0x3ffb34e4 size=0x00000 (     0) load␛[0m
␛[0;32mI (134) esp_image: segment 3: paddr=0x00029f14 vaddr=0x40080000 size=0x00400 (  1024) load␛[0m
␛[0;32mI (144) esp_image: segment 4: paddr=0x0002a31c vaddr=0x40080400 size=0x05cf4 ( 23796) load␛[0m
␛[0;32mI (162) esp_image: segment 5: paddr=0x00030018 vaddr=0x400d0018 size=0x6d504 (447748) map␛[0m
␛[0;32mI (319) esp_image: segment 6: paddr=0x0009d524 vaddr=0x400860f4 size=0x0c554 ( 50516) load␛[0m
␛[0;32mI (340) esp_image: segment 7: paddr=0x000a9a80 vaddr=0x400c0000 size=0x00000 (     0) load␛[0m
␛[0;32mI (341) esp_image: segment 8: paddr=0x000a9a88 vaddr=0x50000000 size=0x00000 (     0) load␛[0m
␛[0;32mI (358) boot: Loaded app from partition at offset 0x10000␛[0m
␛[0;32mI (358) boot: Disabling RNG early entropy source...␛[0m
␛[0;32mI (359) cpu_start: Pro cpu up.␛[0m
␛[0;32mI (362) cpu_start: Starting app cpu, entry point is 0x400810d4␛[0m
␛[0;32mI (0) cpu_start: App cpu up.␛[0m
␛[0;32mI (373) heap_init: Initializing. RAM available for dynamic allocation:␛[0m
␛[0;32mI (380) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM␛[0m
␛[0;32mI (386) heap_init: At 3FFB9528 len 00026AD8 (154 KiB): DRAM␛[0m
␛[0;32mI (392) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM␛[0m
␛[0;32mI (398) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM␛[0m
␛[0;32mI (405) heap_init: At 40092648 len 0000D9B8 (54 KiB): IRAM␛[0m
␛[0;32mI (411) cpu_start: Pro cpu start user code␛[0m
␛[0;32mI (94) cpu_start: Starting scheduler on PRO CPU.␛[0m
␛[0;32mI (0) cpu_start: Starting scheduler on APP CPU.␛[0m
␛[0;31mE (759) sccb: i2c write reg=0x18 err=0xffffffff␛[0m
␛[0;32mI (779) camera_demo: Detected OV2640 camera, using JPEG format␛[0m
␛[0;31mE (819) sccb: i2c write reg=0x54 err=0xffffffff␛[0m
␛[0;31mE (819) sccb: i2c write reg=0x51 err=0x107␛[0m
␛[0;31mE (819) sccb: i2c write reg=0x52 err=0x107␛[0m
␛[0;31mE (819) sccb: i2c write reg=0x55 err=0x107␛[0m
␛[0;31mE (829) sccb: i2c write reg=0x57 err=0x107␛[0m
␛[0;31mE (829) sccb: i2c write reg=0x86 err=0xffffffff␛[0m
␛[0;31mE (839) sccb: i2c write reg=0xd3 err=0xffffffff␛[0m
␛[0;31mE (839) sccb: i2c write reg=0x05 err=0xffffffff␛[0m
␛[0;31mE (849) sccb: i2c write reg=0xe0 err=0xffffffff␛[0m
␛[0;31mE (849) sccb: i2c write reg=0xff err=0xffffffff␛[0m
␛[0;31mE (859) sccb: i2c write reg=0x05 err=0x107␛[0m
␛[0;31mE (889) camera: Failed to set frame size␛[0m
␛[0;31mE (889) camera_demo: Camera init failed with error 0x20002␛[0m
```
Reset helps..

The photo size is 320x240

Zontex, 2020-11-12:
Try to change your USB Cable, I had this issue once and once replaced the cable used for flashing it fixed the problem, sometimes some cables are unstable. if the problem still happening let me know and I will look into it. closing for now.

closed 2020-11-12 by Zontex