m5stack/M5Stack #119: Pipes leave lines behind them in flappy bird example

joeybab3, 2019-01-28:
When playing the flappy bird example, the pipes leave lines behind them as shown below:

[image]

psx88, 2019-03-01:
@joeybab3 Same error on my device. Also the sound seems to be too high pitched and sounds extremely crackling (Tested different sketches and sound files/formats). Getting the grey and the fire version next week. If this issue isn't affecting the next devices, the first one (black edition) is going back to the vendor.

sakabin, 2019-03-04:
Hello, I test is woks well, do you change other things?
[image]

joeybab3, 2019-03-04:
No it’s straight off of GitHub

sakabin, 2019-03-04:
Please try update esp32-arduino and m5stack lib, i test it in different core, they are all woks well

closed 2020-11-12 by Zontex