m5stack/M5Stack #121: Update to current commit of TFT_eSPI, in particular for sprite/PSRAM?

an-erd, 2019-01-31:
I experienced some issues in the context of createSprite() and the use of PSRAM. I saw, that the library [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI) has been updated, in particular to make use of PSRAM for sprites, but the code in M5Stack (`M5Stack/src/utility/*eSPI*` and `Sprite.*`) has not.

Before updating the code in M5Stack to TF_eSPI on my own and start debugging I want to ask whether there is already a plan to update TFT_eSPI short term? 
And, which is the correct branch to use? (master, m5stack-fire or update_eSPI; the latter two are outdated)?

sakabin, 2019-01-31:
sorry, tft lib update maybe need at the end of February, and m5stack-fire,update_eSPI are outdated

an-erd, 2019-01-31:
Ok, I'll go to update the library and to make my checks. 

I saw that the files hadn't been just copied from TFT_eSPI to m5stack, but had been changed, too.

To don't get some known issues, why wasn't the complete library used? Is it a reasonable way to just update what's there in m5stack  with the new version from TF_eSPI?

sakabin, 2019-02-01:
The tft library was not maintained by me early, I will check it after the holiday.

an-erd, 2019-02-01:
I updated the library to the current version of TFT_eSPI, initial tests are fine. The initial issue was not caused by the TFT library (it was too less memory left for WiFi which caused the problem).

If you want me to I can open a pull request for the updated TFT library after some more testing.

tobozo, 2019-02-20:
Please note this is a *breaking change*, after compiling some apps with the latest code repository I observed many anomalies on applications that were working fine 24h ago.

[This commit](https://github.com/m5stack/M5Stack/commit/3168b3edeeeaaa5b988e7873d820b0b3638861c4) seems to kill jpeg rendering and break sprites.

If no fix is planned for this, please don't make a release with this update before clarifying changes or giving time to projet maintainers to adapt.

affected applications examples (worked before merge, broken after):

https://github.com/tobozo/M5Stack-Raytracer
https://github.com/vcraftjp/M5Stack_Particle_demo
https://github.com/drayde/nixietubeM5
https://github.com/PartsandCircuits/M5Stack-Tetris

and there are many more

zhouyangyale, 2019-02-21:
ok，i will check it

zhouyangyale, 2019-04-15:
hi，This code has been modified. Can you try it out there?

tobozo, 2019-04-21:
Hey, after testing a few apps the latest code seems to fix the jpeg decoding :+1

Zontex, 2020-11-12:
Code been modified and now working, closing the issue.

closed 2020-11-12 by Zontex