m5stack/M5Stack #130: [MPU9250] yaw value does not change

thorikawa, 2019-03-14:
I recently purchased M5Stack Gray and tried [MPU9250BasicAHRS.ino](https://github.com/m5stack/M5Stack/blob/master/examples/Modules/MPU9250/MPU9250BasicAHRS/MPU9250BasicAHRS.ino). When I run the sketch and rotate the device, pitch and roll values look correct, but yaw is not. Yaw value keeps almost same value even if I am rotating it. How can I get the correct yaw value? I posted the video here: https://www.youtube.com/watch?v=tkF0MKzGXqA

Any help would be appreciated.

sakabin, 2019-03-14:
Sorry, we also found this problem and I am trying to solve it

thorikawa, 2019-03-15:
I figured it out myself. First of all, the current magnetometer is very sensitive, so it is affected by various conditions including whether USB is inserted or not, whether PC is located nearby the device, etc. Currently magnetometer bias is hard-coded in the [sketch](https://github.com/m5stack/M5Stack/blob/master/examples/Modules/MPU9250/MPU9250BasicAHRS/MPU9250BasicAHRS.ino#L192), but it should be dynamically adjusted because it highly depends on the individual condition.

Also, to make matters worse, M5Stack Gray has powerful neodymium magnets inside the bottom module which also strongly affect the magnetometer.

What I did to solve this issue is that
1. Remove neodymium magnets from the bottom module.
2. Calibrate the magnetometer according to [this article](https://appelsiini.net/2018/calibrate-magnetometer/) and change the hard-coded value.
Also I pass -IMU.mz insted of IMU.mz to [MahonyQuaternionUpdate](https://github.com/m5stack/M5Stack/blob/master/examples/Modules/MPU9250/MPU9250BasicAHRS/MPU9250BasicAHRS.ino#L224) since according to the [document](http://www.invensense.com/wp-content/uploads/2015/02/PS-MPU-9250A-01-v1.1.pdf), MPU9250 magnetometer's z axis is inverse of other sensor's z axis.

I don't send pull request, because as I already said, the bias value depends on the individual device and environment. I think the sketch might as well include dynamic calibration logic and neodymium magnets should be removed especially for this device.

Thank you.

closed 2019-03-18 by zhouyangyale

zhouyangyale, 2019-03-18:
ok，thanks