m5stack/M5Stack #135: Power key usage with IP5306

zavovi, 2019-03-24:
Hi all,
I bought M5stack with Faces and now, I want change behavior of the power key. Is it possible? The best will be only power up from button and then no action and I will handle all from ESP32. 

I found some registers in IP5306 and I tried it. 
1. SYS_CTL1 (reg 0x01), bit7 changed to 1 (long click) - not working
2. SYS_CTL1 (reg 0x01), bit 5 changed to 0 (it was 0 before), it should disable short click for reset, I hope - not working
3. SYS_CTL0 (reg 0x00), bit1 changed to 1, output will still active - not working

Please, could you help me, why it is not working for me? When I reset the device, then both register values are 0x00.

The second question: I read battery level and charging status. It is working, when I plug USB into main module. When I want charge from POGO pins for faces module, then I not see charging status, why? I found, that faces module have the own IP5306, why? Is it a problem? Can I read data from both IP5306?

Thank you very much.

zhouyangyale, 2019-03-25:
Maybe you didn't open iic communication. 
you can use Wire.begin() before the code.

zhouyangyale, 2019-03-25:
You can refer to this now. This version will be online soon.
https://github.com/m5stack/M5Stack/blob/master/src/utility/Power.h
https://github.com/m5stack/M5Stack/blob/master/src/utility/Power.cpp

zavovi, 2019-03-25:
Thank you. The IIC communication is OK, I have my own functions, because I am using esp-idf instead of arduino ide. I can read battery level and status from registers 0x70, 0x71 and 0x78. But when I changed something in register 0x00 and 0x01, then I cannot see any change of behavior. I can try use the code, which you posted. Thank you

zavovi, 2019-03-25:
And what my question about two IP5306 in Faces?

zavovi, 2019-03-25:
OK, I tried some things from your Power files and no change. I tried function "setPowerBoostKeepOn(true)" and I don't understand, what it is doing? The next thing, If I understand right, the function which I want is the "batteryMode(false)", right? I tried it, I read register, I get 0x8A, I changed bit and write it. Then I read register again, and I see right value = 0x0A. Now, I expected no reset when click on power button, but any change. When I click on power button, the device was restarted :-( Why? 
Thank you.

zavovi, 2019-03-25:
Hello, I tried make new read/write functions and better handling all. I tried translate whole IP5306 documentation too. Here is my settings of all registers:

  0.423s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_SYS_CTL0 (0x00) value: 0x30
  0.425s     INFO  [IP5306]      Boost EN: enabled
  0.430s     INFO  [IP5306]      Charger EN: enabled
  0.436s     INFO  [IP5306]      Insert load auto power EN: disabled
  0.443s     INFO  [IP5306]      BOOST output normally open: disabled
  0.450s     INFO  [IP5306]      Key off EN: disabled
  0.457s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_SYS_CTL1 (0x01) value: 0x1d
  0.461s     INFO  [IP5306]      Turn off boost control signal selection: short press twice
  0.473s     INFO  [IP5306]      Switch WLED flashlight control signal selection: short press twice
  0.491s     INFO  [IP5306]      Short press switch boost: disabled
  0.498s     INFO  [IP5306]      Whether to turn on Boost after VIN is pulled out: opened
  0.507s     INFO  [IP5306]      Batlow 3.0V Low Power Shutdown EN: enabled
  0.511s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_SYS_CTL2 (0x02) value: 0x64
  0.532s     INFO  [IP5306]      Light load shutdown time setting: 32s
  0.540s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_CHARG_CTL0 (0x20) value: 0x01
  0.548s     INFO  [IP5306]      Charging full stop setting: 4.17/4.275/4.32/4.365
  0.571s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_CHARG_CTL1 (0x21) value: 0x89
  0.574s     INFO  [IP5306]      Battery end charge current detection: 500mA
  0.581s     INFO  [IP5306]      Charge undervoltage loop setting: 4.55
  0.591s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_CHARG_CTL2 (0x22) value: 0x02
  0.599s     INFO  [IP5306]      Battery voltage setting: 4.2V
  0.611s     INFO  [IP5306]      Constant voltage charging voltage pressurization setting: 28mV
  0.627s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_CHG_DIG_CTL0 (0x24) value: 0xd6
  0.636s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_READ0 (0x70) value: 0x99
  0.644s     INFO  [IP5306]      Charging enable flag: Charging on
  0.651s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_READ1 (0x71) value: 0x70
  0.660s     INFO  [IP5306]      Full of flags: Still charging
  0.661s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_READ2 (0x72) value: 0x61
  0.675s     INFO  [IP5306]      Output light load flag: heavy load
  0.683s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_READ3 (0x77) value: 0x83
  0.690s     INFO  [IP5306]      KEY button double click: 0
  0.697s     INFO  [IP5306]      KEY button long press: 1
  0.703s     INFO  [IP5306]      KEY button short press: 1
  0.710s     INFO  [IP5306]  Register APP_BATTERY_IP5306_REG_LEVEL (0x78) value: 0x00

I tried change "Key off EN" to disable and it seems working. But now, I can disable reset when short click on power button, how?

And I am not understand KEY button register - 0x77. I am getting 0 for click and 1 for long click and for double click, why?
Thank you for you help.

zhouyangyale, 2019-03-26:
Because that button is connected to the reset pin of the microcontroller, 
the function you want cannot be implemented.

zavovi, 2019-03-26:
Oh, it's a pitty :-( 
And I checked connection to second IP5306 into Faces module and it seems, that it is not connected to IIC, right? Then, there is no way, how to get info about charging, when I charging via Faces stand. :-( 

Ok, thank you very much for your help.

zavovi, 2019-03-26:
I am looking into schematic and I see, that power button (PWR) is connected to EN via resistor marked as "R?", is it possible remove this resistor? Which resistor is it on the board? 
Would it work after I remove the resistor?

zhouyangyale, 2019-03-27:
Yes, you can try it. Check if this pin is connected to reset. If so, you can remove the diode.
https://github.com/zhouyangyale/m5-rest/blob/master/rest.png

Zontex, 2020-11-12:
Seems like the questions got the answers, closing for now.

closed 2020-11-12 by Zontex

simoelazzab, 2022-03-19:
Hello, Is it possible to configute the IP5306 to do not automatically enable the output when the it is charging ? When I put my the usb input for charging the output gets enabled, it was expecting it to remain off until the KEY is pressed. Also the KEY does not work when it is charging, I could not turn off the output when the input usb is connected. 

Thanks for help!