m5stack/M5Stack #137: httpd_resp_send_chunk(stream_JPEG)  error

Herwey, 2019-03-26:
Hi
I tried to make the fb process as a task, and send a xSemaphore to httpd_resp_send_chunk once the fb had been processed so that it can send the jpeg stream. But when I open the uri, the web as below:
Content-Type: image/jpeg Content-Length: 5555 ˇÿˇ‡JFIFˇ€C 	 	 $ &&$ #"(-:1(+6+"#2D36;=@A@'0GLF?K:?@>ˇ€C >)#)>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>ˇƒ	ˇƒµ}!1AQa"q2Åë°#B±¡R—$3brÇ %&'()*456789:CDEFGHIJSTUVWXYZcdefghijstuvwxyzÉÑÖÜáàâäíìîïñóòôö¢£§•¶ß®©™≤≥¥µ∂∑∏π∫¬√ƒ≈∆«»…..........................
What should I do ? 
Thanks in advance !

zhouyangyale, 2019-03-27:
hello，I didn't find the demo you said! Is this one?
https://github.com/m5stack/Applications-cam
or this one？
https://github.com/m5stack/m5stack-cam-psram

Herwey, 2019-04-01:
This one: https://github.com/m5stack/m5stack-cam-psram
Sometimes it showed"fmt2rgb888 failed"

Herwey, 2019-04-01:
error as below:
E (14367) esp_jpg_decode: JPG Decompression Failed! Data format error
E (14367) example:http_jpg: fb0 fmt2rgb888 failed

Zontex, 2020-11-12:
Hi, is it still relevant? still having this issue? I'm closing for now, if still relevant feel free to open it again.

closed 2020-11-12 by Zontex