m5stack/M5Stack #140: Is it Possible to Program M5Stack or M5Stick via an Arm device?

mauricecyril, 2019-04-02:
Hi M5Stack team was wondering if it's possible to install the esp32 bored and the M5stack libraries for Arduino IDE or Arduino-CLI IDE on the ARM Architecture (ie. Arm32bit for Raspberry Pi etc, and Arm64bit for Arm8 devices like mobile phones etc).

Could you please post instructions of how to do this so that users who would like to program the devices using their mobile phones or in SBC can do so either via the command line or using the Arduino IDE?

zhouyangyale, 2019-04-08:
We have not tried

mauricecyril, 2019-04-15:
Hi zhouyangyale, i tried to install the libraries on an Arm device using arduino for arm and am getting errors installing the ESP32 and m5stack libraries.

Zontex, 2020-11-12:
Raspberry Pi (an arm based device) can program ESP32 and Arduino devices so I believe although we haven't tried that M5Stack device will work that way as well. regarding the use of mobile phone or anything else (not a raspberry pi) i'm not sure it can be done.

Here is article on Arduino website how to program using the raspberry pi: https://create.arduino.cc/projecthub/techno_z/program-your-arduino-from-your-raspberry-pi-3407d4

this is not an issue so I'm closing for now.

closed 2020-11-12 by Zontex