m5stack/M5Stack #142: PWM control of the backlight emits unpleasant 10 kHz noise

lovyan03, 2019-04-04:
Currently, 10KHz is specified for PWM control of backlight in M5Display.cpp.

https://github.com/m5stack/M5Stack/blob/master/src/M5Display.cpp#L13

It turned out that this is the cause of ESP32 emitting an offensive treble.
Also, I confirmed that when I put an FFT on the M5GoBase's microphone input, a response at 10 KHz came out.

As a trial, I changed this to a sufficiently low value (120 Hz) or a sufficiently high value (44100 Hz) and succeeded in turning off the offensive treble.

I would like to change this 10 kHz value, but will that cause other problems?

sakabin, 2019-04-08:
Maybe not, we will test it, thanks

zhouyangyale, 2019-04-12:
We have reduced the noise, but there is no way to completely eliminate it. 
Can be due to hardware circuit reasons

lovyan03, 2019-04-12:
Yes. However, I think that it is possible to set outside the range of human hearing.

lovyan03, 2019-04-12:
The noise of PWM is not heard from the speaker.
It is emitted directly from ESP32.
Adjustments to the speakers are meaningless.

sakabin, 2019-04-12:
We found that sometimes if used 120 HZ will make screen flicker, so suggest use 44100 HZ

lovyan03, 2019-04-17:
I made a pull request. #147 

If the meaning is not clear, execute the following code, put the M5Stack on your ear and press the button to hear the change in mosquito noise.
```
#include <M5Stack.h>
void setup() {
  M5.begin();
}

uint32_t Hz = 10000;
void loop() {
  delay(1);
  M5.update();
  if (M5.BtnB.isPressed() && 0 < Hz) {
    ledcWriteTone(7, --Hz);
  }
  if (M5.BtnC.isPressed() && 44100 > Hz) {
    ledcWriteTone(7, ++Hz);
  }
  M5.Lcd.setCursor(100,100);
  M5.Lcd.printf("%7d", Hz);
}
```

closed 2019-04-17 by lovyan03