m5stack/M5Stack #148: M5Faces turns off when display brightness set to 20

zsoltmazlo, 2019-04-20:
Hi!

This issue is a very nasty one and only occurs in a special circumstances.

I'm using an M5Faces device with the Charge Base and I wanted to achieve that if I don't touch the device than it decreases the brightness of the screen and show special graphics. All of this are works well when the device is plugged in via USB cabel or even when running from battery, but when I place it to the charge base, after some time (~40s) switches off.

If I remove the brightness change, everything works fine.

I can't even decide if it is a hardware or a software bug, can anybody help me out?

zsoltmazlo, 2019-04-30:
After some research, I have a clue on this issue: according to [this forum entry](http://forum.m5stack.com/topic/62/ip5306-automatic-standby), when power consumption of the device goes under 45mA, after 32 seconds the IP5306 chip automatically goes to standby mode and did not power the unit. I do not understand chinese, but using google translate and checking the [datasheet](https://www.electrodragon.com/w/images/f/f0/IP5306_datasheet_v1.01.pdf) there is an entry in it about this functionality. Can you confirm this and show me how can I disable this in the chip?

nmori, 2019-05-01:
According to the data sheet, if the current is low, the power will automatically shut off after a certain time.
You can probably disable this feature with the Power function. 
`M5.Power.setLowPowerShutdown (false);`

zsoltmazlo, 2019-05-01:
Last night I was found out that the M5Stack library had an update 2 weeks ago and that release contains all of these power related setup. So I was able to use it and now the device does not shuts down on low power usage. 

Although it is reasonable what @nmori says, that was not the solution: I could only guest why, but these lines were the fixes for my trouble:

`M5.Power.setPowerBoostSet(true);
 M5.Power.setPowerBoostKeepOn(true);`

closed 2019-05-01 by zsoltmazlo