m5stack/M5Stack #149: M5Stack Library 0.2.6 with Fire wont play M5 speaker functions

rickmini, 2019-04-21:
M5 speaker functions for beep and tone will not work with 0.2.6. Reverted to 0.2.5 to get them to work again.

lovyan03, 2019-04-22:
Perhaps that is due to this change.

https://github.com/m5stack/M5Stack/commit/3168b3edeeeaaa5b988e7873d820b0b3638861c4#diff-ef038e34f2e1a37eed66b42423345c6eR35

You need this code in setup.
  M5.Speaker.begin();

rickmini, 2019-04-22:
That solved the problem. I assumed that only M5.begin() was required for all functions to work. What are other initializations that I should know about other than M5.Speaker.begin()?

zhouyangyale, 2019-04-24:
The M5.Speaker.begin() function was originally included in M5.begin(), but was removed in version 0.2.6.

closed 2019-04-24 by rickmini