m5stack/M5Stack #150: Fire MPU9250 Sample will not work

rickmini, 2019-04-21:
Both library versions of 0.2.5 and 0.2.6 will not work with Examples-Fire-MPU9250. Initial values are printed to the screen. 
But they never update. The loop continues properly but the values remain the same.
I tried by selecting the board as FIRE and also as M5 Core, both produce the same issue.

lovyan03, 2019-04-22:
For I2C communication, the Arduino-ESP32 library version is involved and it does not work with the old one.
What version number is Arduino-ESP32 ?

rickmini, 2019-04-22:
im using 1.02.rc
when i tried 1.02 (just released) the Fire speaker functions stopped working

lovyan03, 2019-04-26:
I also tried. It looks like it is not working properly.  

This has worked.  
https://github.com/m5stack/M5Stack/blob/master/examples/Modules/MPU9250/MPU9250BasicAHRS/MPU9250BasicAHRS.ino

lovyan03, 2019-04-27:
I confirmed that the sample of MPU9250 does not work correctly with the change of 2019/04/12.
I try fix it.

nmori, 2019-04-27:
Incorrect rewriting when refactoring. As a result, I2C sub-funciton stopped working after 2019/4/12. 
 I post a pull-request to the modification to #154 . I'm sorry for the mistake

rickmini, 2019-04-30:
I saw that a new commit was made to the m5stack library with your changes. I deleted my old lib, and added the new .zip lib. Now nothing works. I get a compile error that says: 
Invalid library found in C:\Users\rickm\Documents\Arduino\sketch\libraries\M5Stack: no headers files (.h) found in C:\Users\rickm\Documents\Arduino\sketch\libraries\M5Stack

nmori, 2019-04-30:
The placement of the M5stack library is assumed to be placed in the library folder above it, not a under folder of sketch. I tested it in library folder. I want you to try it once in that arrangement.

rickmini, 2019-05-01:
It was my issue with the library placement. The new library works fine now. The MPU9250 is responding properly. Thanks to you and lovyan03.

closed 2019-05-02 by EeeeBin

AndrewCapon, 2019-09-19:
Hi,

I have downloaded the master branch as a zip, removed the pre-existing M5stack library which I downloaded via the Arduino library manager and added the downloaded zip as a library into the Arduino IDE.

With the both the two samples I do not get relevant data, just a static set:

1. basics->IMU with #define M5STACK_MPU9250 
2. Fire->MSStackFire_MPU9520

Has anyone any ideas?

Many thanks

Andy