m5stack/M5Stack #151: TFT_Flash_Bitmap example is not working with M5Stack Library 0.2.6

hardmax, 2019-04-23:
With the new library 0.2.6 TFT_Flash_Bitmap example is not work.

Arduino compile and upload, but the image is not appear in screen....don't show or appear errors, only images is not showing

I use new arduino ESP32 library.

When I change M5Stack library at 0.2.5, the images work againg.

zhouyangyale, 2019-04-24:
Which function are you using?

hardmax, 2019-04-24:
I just try to use the example as it is.....what you mean function??..the example use:

M5.Lcd.setWindow and
M5.Lcd.pushColors

lovyan03, 2019-04-24:
M5.Lcd is made by copying TFT-eSPI, but the examples has not been corrected although the function has changed largely due to the change made in February this year.
Also, immediately after that, TFT-eSPI has many bug fixes.
Therefore, I think that it is necessary to make In_eSPI again based on the latest TFT-eSPI and correct the examples as well.

The current stable version is 0.2.5...

hardmax, 2019-04-24:
Ok...I review and make the changes and now its working:

// Icons are stored in tabs ^ e.g. Alert.h etc above this line
// more than one icon can be in a header file

/*
 This sketch demonstrates loading images from arrays stored in program (FLASH) memory.

 Works with TFT_eSPI library here:
 https://github.com/Bodmer/TFT_eSPI

 This sketch does not use/need any fonts at all...

 Code derived from ILI9341_due example

 Make sure all the display driver and pin comnenctions are correct by
 editting the User_Setup.h file in the TFT_eSPI library folder.

 #########################################################################
 ###### DON'T FORGET TO UPDATE THE User_Setup.h FILE IN THE LIBRARY ######
 #########################################################################
*/

#include <M5Stack.h>

// Include the header files that contain the icons
#include "alert.h"
#include "Close.h"
#include "Info.h"

long count = 0; // Loop count

void setup()
{
  M5.begin();

  M5.Lcd.fillScreen(TFT_BLACK);
  M5.Lcd.setSwapBytes(true);
  
  // Draw the icons
  M5.Lcd.pushImage(100, 100, infoWidth, infoHeight, info);
  M5.Lcd.pushImage(140, 100, alertWidth, alertHeight, alert);
  M5.Lcd.pushImage(180, 100, closeWidth, closeHeight, _close);

  // Pause here to admire the icons!
  delay(2000);

}

void loop()
{
  // Loop filling and clearing screen
  M5.Lcd.pushImage(random(M5.Lcd.width() -  infoWidth), random(M5.Lcd.height() -  infoHeight),  infoWidth,  infoHeight, info);
  M5.Lcd.pushImage(random(M5.Lcd.width() - alertWidth), random(M5.Lcd.height() - alertHeight), alertWidth, alertHeight, alert);
  M5.Lcd.pushImage(random(M5.Lcd.width() - closeWidth), random(M5.Lcd.height() - closeHeight), alertWidth, closeHeight, _close);
  
  // Clear screen after 100 x 3 = 300 icons drawn
  if (100 == count++) {
    count = 1;
    M5.Lcd.setRotation(2 * random(2)); // Rotate randomly to clear display left>right or right>left to reduce monotony!
    M5.Lcd.fillScreen(TFT_BLACK);
    M5.Lcd.setRotation(1);
    //Serial.println(millis());
  }
}

lovyan03, 2019-04-27:
Fix completed. ( #152 )
Please wait for the merge.

lovyan03, 2019-05-18:
Probably, this issue is solved in version 0.2.7

Zontex, 2020-11-12:
Issue solved, closing.

closed 2020-11-12 by Zontex