m5stack/M5Stack #155: Sprite is not working properly

mongonta0716, 2019-04-29:
16bitSprite is not working properly with the change of 0.2.6. It can only display up to 237x237.
It is probably working in 0.2.5.

[image]

EeeeBin, 2019-04-29:
The tft library has been updated, please try again

mongonta0716, 2019-04-29:
Hi,@EeeeBin 

I tried updating but the problem is not changed. Is there this problem in your environment?

Thanks.

EeeeBin, 2019-04-29:
Could you share your code, I want to test it tomorrow

mongonta0716, 2019-04-29:
I see.

This App is M5Stack-Avatar example 「face-and-color」.
When button A is pressed, it switches to dog face and the phenomenon occurs.

https://github.com/meganetaaan/m5stack-avatar

Thanks.

mongonta0716, 2019-04-29:
I further investigated the cause, it seems to occur when you set Sprite's Color Depth to 1 bit instead of 16 bit
If you reproduce with face-and-color, please investigate.

mongonta0716, 2019-04-30:
I continued to investigate. It also depends on the type of M5Stack.

●Condition that causes the issue
`Type | 1b | 8b | 16b |`
`Fire  | NG | OK   | OK   |`
`Gray | NG | OK   | NG   |`

lovyan03, 2019-04-30:
I fixed it #157

mongonta0716, 2019-04-30:
It worked fine.　#158

Thanks!