m5stack/M5Stack #156: Wrong bit being set

jtgrassie, 2019-04-29:
https://github.com/m5stack/M5Stack/blob/4b7b423b302cc7382280c826f089d7aa09cf9142/src/utility/Power.cpp#L99

This is setting the boost enable bit (5), not the auto load bit (2).

Reference sheet [here](https://docviewer.yandex.ru/view/0/?*=aciD0ZpRNaSTXGfd3E2r30McaxV7InVybCI6InlhLWRpc2stcHVibGljOi8vZTlnMElGZVlCM3ZwaTlxbklETy9wTThHZzRXOVhCc1F4UmRqNHZqYzNxVEJtay9COU1EVkFpV1BUTS9wVFR6QnEvSjZicG1SeU9Kb25UM1ZvWG5EYWc9PSIsInRpdGxlIjoiSVA1MzA2IEkyQyByZWdpc3RlcnMucGRmIiwidWlkIjoiMCIsInl1IjoiOTQzOTEwODMxMTU1NTYwOTI5MyIsIm5vaWZyYW1lIjpmYWxzZSwidHMiOjE1NTY1NjU0MjA4ODJ9).

lovyan03, 2019-04-29:
LightLoad ...  
When deep sleep, IP5306 turns off power after a certain period of time.  
(It means that you can not wake up from deep sleep with a button.)  
  
AutoLoad ...  
When poweroff, IP5306 turns on power after connecting a load that consumes power.  
  
The IP5306 is an IC that originally controls mobile batteries.  
Therefore, there is a function to control the power supply depending on the connection state of the load.

lovyan03, 2019-04-29:
Sorry,  I understand it,  
it's bug.

jtgrassie, 2019-04-30:
You should be toggling bit 2 not bit 5 to toggle AutoLoad and there doesn't seem to be a way for you to implement `POWER::setKeepLightLoad` directly on the IP5306 (e.g. you cannot _write_ to `REG_READ2` with the `LIGHT_LOAD_BIT`). You'd have to do something else to keep a minimal load on the circuit which prevented the IC from powering off. You should probably also implement a method to change the light load shutdown time setting. Right now, if you want to deep sleep a device, as soon as 30 seconds has passed, you cant wake - which is a serious issue for devices intended to be run for a long period on battery power.

EeeeBin, 2019-04-30:
you are right,if call  ```M5.Power.setKeepLightLoad(true)```, IP5306 maybe can`t power on if only use battery, ```M5.Power.setPowerBoostKeepOn(true) ```can instead it
The datasheet mean if bit 5 set 0, IP5306 can't automatically shut down if current is too small, I am looking for why it doesn't work
[image]

EeeeBin, 2019-04-30:
if call ```M5.Power.setPowerBoostKeepOn(true)``` mean double click power btn will not work, but if esp32 go to deep sleep, it can be wake after 30 seconds

jtgrassie, 2019-04-30:
I actually tested a few things and no matter what `setPowerBoostKeepOn` was set to (0 or 1), when the device was sent to a deep sleep, on battery, after 30 seconds there was no way to get it to wake except for plugging it in.

According to the docs, boost enabled just means keep the step-up DCDC converter switched on (and this means keep it at 5V 1A output). It does not mean switch on/off the smart load detector (which automatically switches the IC to standby mode - which the only way to wake from is supply load it seems). Therefore:

> if call M5.Power.setPowerBoostKeepOn(true) mean double click power btn will not work, but if esp32 go to deep sleep, it can be wake after 30 seconds

Is not correct. Once the IC has gone into standby (deepSleep device for more than 30 seconds), you cannot wake it without connecting power. Such a shame this IP5306 was chosen for the M5.

EeeeBin, 2019-04-30:
Please try sleep example, just updated，it will be ok if your M5 IP5306 can be control

jtgrassie, 2019-04-30:
It doesn't fix the problem. Example:

```
void setup() {
  M5.begin();
  M5.Power.begin();
  M5.Lcd.begin();
  M5.Lcd.fillScreen(BLACK);
  M5.Lcd.setCursor(0, 0);
  M5.Lcd.setTextColor(GREEN, BLACK);
  M5.Lcd.setTextSize(1);
}
void loop() {
  M5.Lcd.printf("sleep for 60 seconds\n");
  delay(2500);
  M5.Power.deepSleep(60000000);
  M5.Lcd.printf("never get here\n"); // no buttons, nothing, will get here unless you plug in power.
}
```

nmori, 2019-04-30:
1) This problem occurred during the Power class creation and refactoring process. Although setKeepLightLoad () is not working, you can achieve the desired behavior of maintaining power using setPowerBoostKeepOn () in a limited way. The expected behavior when creating a class is synonymous. Therefore, I propose to abolish the setKeepLightLoad function and unify it to setPowerBoostKeepOn.
2) problem of return from sleep. It does not occur when the power is supplied from USB. Problems can occur if you remove USB and power off the IP 5306. LightSleep can return, but DeepSleep can not return after a certain period of time. As described in other comments, IP 5306 shuts down at low current and ESP 32 turns off.
3) From the above, this function does not always work in the desired way, but this is due to hardware limitations. Therefore, I would like to take measures to describe the document in the future.

jtgrassie, 2019-04-30:
Precisely. You could work around the hardware limitation however. If you implemented something for the IP5306 whereby you switched a load of 45mA to the IC in setKeepLightLoad when entering deep sleep, that would be a fairly simple hardware mod. Otherwise start using a better battery/charge IC.

The problem with setPowerBoostKeepOn & LightSleep is that it defeats the purpose of entering a sleep mode (reducing battery consumption). Therefore the only current workaround is to enter deep sleep for < 30 secs at a time.

nmori, 2019-04-30:
In order to realize the purpose of not wanting to consume power, I understood that it is better not to return until the specified time while repeating deepSleep.

The purpose of the LightSleep implementation is to reduce the power without resetting the CPU and to meet the needs to continue programming from the function-call point after sleep.

So I think that different functions are used depending on whether you put in a function that gives priority to power, or give priority to keeping the state while saving energy there.

In any case, it will not work if the IP5306 shutdown time is longer than the specified number of seconds. If you think about the point that can be extended to 64 seconds, it will be more effective if you add a function to extend the number of seconds, so I will write and check the source from now on.

EeeeBin, 2019-04-30:
Test your code in M5Stack Fire, it work well, Wake up every 60 seconds or press btnA wake up, So please make sure your IP5306 or Fire is fine

nmori, 2019-04-30:
I created an addition of code that could extend the time.
With this program, even if you unplug the USB, it can automatically recover for up to 63 seconds. If you use an external power supply such as USB, you can recover without this code.
https://github.com/nmori/M5Stack/blob/master/src/utility/Power.cpp

nmori, 2019-04-30:
The reason why the previous sleep sketch did not move correctly is because I2C is not initialized and the correct instruction has not been created.

Because the new sketch is properly initialized, the steps to disable the Auto-off feature have been confirmed to work correctly.

The instruction disables Sharddown while IP 5306 continues to run, so it consumes some power, but the sleep operation can recover for more than 64 seconds.

I understood the operation mode of IP5306.

Thank you for the information.

jtgrassie, 2019-04-30:
> Test your code in M5Stack Fire, it work well, Wake up every 60 seconds or press btnA wake up, So please make sure your IP5306 or Fire is fine

Nope, even with your latest edits, I can get back from deep sleep < 60 seconds, but not if I set to 120 seconds. You cannot disable auto shutdown 5306 completely. 

> The instruction disables Sharddown while IP 5306 continues to run, so it consumes some power, but the sleep operation can recover for more than 64 seconds.

Not on battery only from deep sleep.

nmori, 2019-04-30:
In the old model that IP5306 is not correspondence, I was able to confirm the phenomenon that cannot be restarted after 32 seconds.

However, it does not happen when powered from USB.

In the new hard, tried a lot of tests, but the actual movement and the movement in the specifications do not match. The new library restarts correctly after 120 seconds.

I measured the current of the USB as I thought that the IP5306 is flowing a lot of power, but it does not flow.

I can see that it's working, but I can't explain it to the extent I understand it. Is the specification of the hard or IP5306 changed?

[image]

jtgrassie, 2019-04-30:
> However, it does not happen when powered from USB.

Sure. But the whole point of deep sleep is to preserve *battery*. So saying it works when powered by USB is not helpful at all.

> In the new hard, tried a lot of tests, but the actual movement and the movement in the specifications do not match. The new library restarts correctly after 120 seconds.

With you current master, the device does NOT restart from deep sleep after 120 seconds if you are only on battery. Any test with the USB plugged in is pointless. You use deep sleep for a battery powered device to conserve batter power.

nmori, 2019-04-30:
According to the current story, running the M5stack in battery, deep sleep mode, low current, will shut down the IP5306, the only way to stop it is to increase the current.

It is a video that was confirmed in the "USB power off state" that was insufficient in the previous verification.

[testmovie.zip] (https://github.com/m5stack/M5Stack/files/3132064/testmovie.zip)

In this video, you can wake up every two minutes while the battery is running.

Therefore, the current situation is not in line with the story.

However, with the information on hand, the reason can not be reached. The sketches I'm testing are below.

`#include <M5Stack.h>

void setup() {
  M5.begin();
  M5.Power.begin();
  M5.Lcd.setBrightness(200);

  M5.Power.setWakeupButton(BUTTON_A_PIN);
  if(!M5.Power.canControl()) {
    M5.Lcd.printf("IP5306 is not i2c version\n");
  }
}

void loop() {

  M5.update();
  bool c=M5.Power.isResetbyPowerSW();
  bool d=M5.Power.isResetbyDeepsleep();
 
  M5.Lcd.printf("<<Sleep test>>\n");
  M5.Lcd.printf("power-on triggered at:%s%s\n\n",c?"POWER-SW":"",d?"DeepSleep-end":"");

  M5.Lcd.printf("\n\nGo deepSleep (5s or press buttonA) ");
  delay(2500);
  // call this fun, power btn will Invalid
  M5.Power.deepSleep(120000000);
}`

Test hardware : M5Stack Fire (2018.2A)
I want to know if I can reproduce the problem on another user's M5stack

jtgrassie, 2019-04-30:
We've actually got quite a few M5 Fires and all of them are working the same. On battery, deep-sleep, never wakes.

Using light-sleep is ok, just not deep-sleep.

jtgrassie, 2019-04-30:
I can get waking from deep sleep > 60s by calling `M5.Power.setLowPowerShutdown(false);`

closed 2019-05-02 by EeeeBin

hardmax, 2019-07-24:
Hi

I have problems to that too,  I see that some units don't power on when are USB conected, i have to push reset button to M5 power on....some units power on automatic, others not (Had to push reset button)...In UNits that power on when are power it by USB, is working deepSleep a another functions, in the other not (I buy 50 units and 20 has that problem).

I do a reflow in units that not power On and some them are worked againg the automatic power On USB connect, a other still the problem.

I check all M5.power.set instruccions, but nothing happen with that units.....I thing that its a problem in IP5306 that not recive good information because are badly welded..

20 units from 50 has the problem and can't uses it in places without supervision. (If they are power off, don't come back before some body press reset button...they dont power on if energy come back)

jbrepogmailcom, 2020-01-22:
This code works well, ESP32 can be waken up anytime, I wake it by interrupt:

    #define IP5306_ADDR 0x75
    #define IP5306_REG_SYS_CTL0 0x00
    #define IP5306_REG_SYS_CTL1 0x01
    #define IP5306_REG_SYS_CTL2 0x02
    Wire.beginTransmission(IP5306_ADDR);
    Wire.write(IP5306_REG_SYS_CTL0);
    //Wire.write(0b110111);
    Wire.write(55);
    // 0x37 = 0b110111 TCALL example
    /*
    [1] Boost EN (default 1) [EXM note: if 0 ESP32 will not boot from battery]
    [1] Charger EN (1) [EXM note: did not observe difference]
    [1] Reserved (1) [EXM note: did not observe difference]
    [1] Insert load auto power EN (1) [EXM note: did not observe difference]
    [1] BOOST output normally open ( 1) [EXM note: if 0 will shutdown when
    ESP32 sleeps after 32s]
    [1] Key off EN: (0) [EXM note: could not detect difference]
    */ 
    Wire.write(IP5306_REG_SYS_CTL1);
    Wire.write(29);
    Wire.write(IP5306_REG_SYS_CTL2);
    Wire.write(100);
    //Wire.write(0x1D); // Set HEX:1D DEC:29 BIN:11111
    /*
    [1] Turn off boost control signal selection: short press twice
    [1] Switch WLED flashlight control signal selection: short press twice
    [1] Short press switch boost: disabled
    [0] Whether to turn on Boost after VIN is pulled out: opened
    [1] Batlow 3.0V Low Power Shutdown EN: enabled
    */