m5stack/M5Stack #164: can't compile on macos

Keoni80, 2019-05-17:
trying to compile the basic example and always have the same error:

/Users/juan/Documents/Arduino/libraries/M5Stack/src/utility/Power.cpp: In member function 'void POWER::powerOFF()':
/Users/juan/Documents/Arduino/libraries/M5Stack/src/utility/Power.cpp:360:35: error: 'ESP_SLEEP_WAKEUP_ALL' was not declared in this scope
   esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);

                                   ^
/Users/juan/Documents/Arduino/libraries/M5Stack/src/utility/Power.cpp:361:28: error: 'gpio_deep_sleep_hold_dis' was not declared in this scope
   gpio_deep_sleep_hold_dis();

                            ^
Se encontraron múltiples librerías para "SD.h"
Usado: /Users/juan/Documents/Arduino/hardware/espressif/esp32/libraries/SD
 No usado: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
Error compilando para la tarjeta M5Stack-Core-ESP32.

it works fine on windows :S

lovyan03, 2019-05-17:
Probably, need update ArduinoESP32 (v1.0.2)

Keoni80, 2019-05-20:
how do i do that?

Keoni80, 2019-05-20:
done, i deleted the main folder, and then reinstall , thanks

brentru, 2019-05-20:
Having this issue as well:

```
/Users/brentrubell/Documents/Arduino/libraries/M5Stack/src/utility/Power.cpp: In member function 'void POWER::powerOFF()':
/Users/brentrubell/Documents/Arduino/libraries/M5Stack/src/utility/Power.cpp:360:35: error: 'ESP_SLEEP_WAKEUP_ALL' was not declared in this scope
   esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);

                                   ^
/Users/brentrubell/Documents/Arduino/libraries/M5Stack/src/utility/Power.cpp:361:28: error: 'gpio_deep_sleep_hold_dis' was not declared in this scope
   gpio_deep_sleep_hold_dis();

                            ^
Multiple libraries were found for "WiFi.h"
 Used: /Users/brentrubell/Documents/Arduino/hardware/espressif/esp32/libraries/WiFi
 Not used: /Users/brentrubell/Documents/Arduino/libraries/WiFiNINA
 Not used: /Applications/Arduino.app/Contents/Java/libraries/WiFi
Multiple libraries were found for "SD.h"
 Used: /Users/brentrubell/Documents/Arduino/hardware/espressif/esp32/libraries/SD
 Not used: /Users/brentrubell/Documents/Arduino/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
```

lovyan03, 2019-05-21:
Maybe, that is not the latest (arduino-esp32 v1.0.2 release).

[image]

brentru, 2019-05-22:
@lovyan03 Looks like it didn't release properly from them, works OK now. Feel free to close!

rodlenz, 2019-05-24:
Dear @Keoni80 I`m facing the exactly same issue, do you mind to explain in more details what exqactly you did to solve this problem? Tks in advance for the additional help.

Keoni80, 2019-05-24:
@rodlenz i deleted the ESP32 folder, and downloaded again with terminal, then it works with arduino IDE, but still doesn't work with PlatformIO

rayozzie, 2019-06-19:
macOS Mojave, Arduino 1.8.9, esp32 1.0.2, Examples/M5Stack/Basics/Display, error compiling for board M5Stack-Core-ESP32:

`Arduino: 1.8.9 (Mac OS X), Board: "M5Stack-Core-ESP32, QIO, 80MHz, Default, 921600, None"

Build options changed, rebuilding all
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setKey(char*, char*, char*)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:109:41: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         loraDebugPrint(DEFAULT_DEBUGTIME);
                                         ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:120:41: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         loraDebugPrint(DEFAULT_DEBUGTIME);
                                         ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:131:41: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         loraDebugPrint(DEFAULT_DEBUGTIME);
                                         ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'bool LoRaWanClass::setDataRate(_data_rate_t, _physical_type_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:144:38: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
       debugPrint("Unknown datarate\n");
                                      ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:159:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:167:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:140:17: warning: unused variable 'str' [-Wunused-variable]
     const char *str;
                 ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setPower(short int)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:180:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setPort(unsigned char)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:193:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setAdaptiveDataRate(bool)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:203:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::getChannel()':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:212:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setChannel(unsigned char, float)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:230:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setChannel(unsigned char, float, _data_rate_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:245:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setChannel(unsigned char, float, _data_rate_t, _data_rate_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:260:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setUnconfirmedMessageRepeatTime(unsigned char)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:562:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setConfirmedMessageRetryTime(unsigned char)':
[... cut ...]
 Not used: /Users/rozzie/Documents/Arduino/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/M5Display.cpp: In member function 'void M5Display::drawBmpFile(fs::FS&, const char*, uint16_t, uint16_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/M5Display.cpp:127:23: warning: unused variable 'col' [-Wunused-variable]
   uint16_t w, h, row, col;
                       ^
Error compiling for board M5Stack-Core-ESP32.

`

Zontex, 2020-11-11:
Seems the problem has been solved, closing the issue.

closed 2020-11-11 by Zontex