m5stack/M5Stack #167: Untested library

pmurakh, 2019-06-07:
Using new M5stack Fire, trying to compile any example sketch in PlatformIO or Arduino fails with errors from Power.cpp lines 361 and 360. Function not defined.
Removing the 2 lines allows sketches to compile in Arduino IDE. Same error exists in github library or library installed from Arduino IDE.

In Arduino IDE attempting the MPU example sketch compiles and uploads but nothing is displayed on the Fire Screen.
Please test released libraries!!!

EeeeBin, 2019-06-08:
In arduino ide, try update your arduino-esp32 lib to Newest, see https://github.com/m5stack/M5Stack/issues/164
maybe PlatformIO did not update the latest esp32, try use 0.2.5 version m5stack lib

pmurakh, 2019-06-08:
Thankyou. Compiles now in Arduino IDE and MPU example works (from Arduino IDE).

closed 2019-06-08 by pmurakh