m5stack/M5Stack #177: M5Stack on Arduino unusable when Preferences / Compiler Warnings == All

rayozzie, 2019-07-11:
I'm developing my own arduino libraries for 3rd parties, and so (as is standard practice) I must run the Arduino IDE with the "Compiler Warnings" setting in Arduino Preferences set to "All", so that my code doesn't contain warnings that would confuse or block a customer.

The examples for my library are built using M5Stack, which is simply wonderful.  Thank you.

However, I am blocked from using the M5Stack libraries when compiling with Compiler Warnings set to "All".  

The presence of some of these warnings is just annoying, but the compiler treats some of these warnings as errors and will not proceed. 

If you can fix these it would be greatly appreciated.

(I'm using Arduino 1.8.9 on MacOS Mojave.)

********************

/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setKey(char*, char*, char*)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:109:41: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         loraDebugPrint(DEFAULT_DEBUGTIME);
                                         ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:120:41: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         loraDebugPrint(DEFAULT_DEBUGTIME);
                                         ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:131:41: warning: large integer implicitly truncated to unsigned type [-Woverflow]
         loraDebugPrint(DEFAULT_DEBUGTIME);
                                         ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'bool LoRaWanClass::setDataRate(_data_rate_t, _physical_type_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:144:38: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
       debugPrint("Unknown datarate\n");
                                      ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:159:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:167:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:140:17: warning: unused variable 'str' [-Wunused-variable]
     const char *str;
                 ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setPower(short int)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:180:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setPort(unsigned char)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:193:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setAdaptiveDataRate(bool)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:203:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::getChannel()':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:212:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setChannel(unsigned char, float)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:230:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setChannel(unsigned char, float, _data_rate_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:245:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setChannel(unsigned char, float, _data_rate_t, _data_rate_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:260:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setUnconfirmedMessageRepeatTime(unsigned char)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:562:37: warning: large integer implicitly truncated to unsigned type [-Woverflow]
     loraDebugPrint(DEFAULT_DEBUGTIME);
                                     ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'void LoRaWanClass::setConfirmedMessageRetryTime(unsigned char)':
[... 50 lines cut ...]
 ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'short int LoRaWanClass::receivePacket(char*, short int, short int*)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:490:43: error: 'tmp' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                 result = result * 16 + tmp;
                                           ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:448:43: error: 'tmp' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                 result = result * 16 + tmp;
                                           ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp: In member function 'short int LoRaWanClass::receivePacketP2PMode(unsigned char*, short int, short int*, unsigned char)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/LoRaWan.cpp:875:43: error: 'tmp' may be used uninitialized in this function [-Werror=maybe-uninitialized]
                 result = result * 16 + tmp;
                                           ^
cc1plus: some warnings being treated as errors
Multiple libraries were found for "SD.h"
 Used: /Users/rozzie/Library/Arduino15/packages/esp32/hardware/esp32/1.0.2/libraries/SD
 Not used: /Users/rozzie/Documents/Arduino/libraries/SD
 Not used: /Applications/Arduino.app/Contents/Java/libraries/SD
exit status 1
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/M5Display.cpp: In member function 'void M5Display::drawBmpFile(fs::FS&, const char*, uint16_t, uint16_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/M5Display.cpp:127:23: warning: unused variable 'col' [-Wunused-variable]
   uint16_t w, h, row, col;
                       ^
Error compiling for board M5Stack-Core-ESP32.

********************

rayozzie, 2019-07-11:
More:

Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp: In member function 'void MPU9250::readAccelData(int16_t*)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:64:3: warning: 'void MPU9250::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*)' is deprecated [-Wdeprecated-declarations]
   readBytes(MPU9250_ADDRESS, ACCEL_XOUT_H, 6, &rawData[0]);  // Read the six raw data registers into data array
   ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:250:12: note: declared here
       void readBytes(uint8_t, uint8_t, uint8_t, uint8_t *)   __attribute__ ((deprecated));
            ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:64:58: warning: 'void MPU9250::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*)' is deprecated [-Wdeprecated-declarations]
   readBytes(MPU9250_ADDRESS, ACCEL_XOUT_H, 6, &rawData[0]);  // Read the six raw data registers into data array
                                                          ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:250:12: note: declared here
       void readBytes(uint8_t, uint8_t, uint8_t, uint8_t *)   __attribute__ ((deprecated));
            ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp: In member function 'void MPU9250::readGyroData(int16_t*)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:72:3: warning: 'void MPU9250::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*)' is deprecated [-Wdeprecated-declarations]
   readBytes(MPU9250_ADDRESS, GYRO_XOUT_H, 6, &rawData[0]);  // Read the six raw data registers sequentially into data array
   ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:250:12: note: declared here
       void readBytes(uint8_t, uint8_t, uint8_t, uint8_t *)   __attribute__ ((deprecated));
            ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:72:57: warning: 'void MPU9250::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*)' is deprecated [-Wdeprecated-declarations]
   readBytes(MPU9250_ADDRESS, GYRO_XOUT_H, 6, &rawData[0]);  // Read the six raw data registers sequentially into data array
                                                         ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:250:12: note: declared here
       void readBytes(uint8_t, uint8_t, uint8_t, uint8_t *)   __attribute__ ((deprecated));
            ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp: In member function 'void MPU9250::readMagData(int16_t*)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:83:6: warning: 'uint8_t MPU9250::readByte(uint8_t, uint8_t)' is deprecated [-Wdeprecated-declarations]
   if(readByte(AK8963_ADDRESS, AK8963_ST1) & 0x01) {
      ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:249:15: note: declared here
       uint8_t readByte(uint8_t, uint8_t)   __attribute__ ((deprecated));
               ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:83:41: warning: 'uint8_t MPU9250::readByte(uint8_t, uint8_t)' is deprecated [-Wdeprecated-declarations]
   if(readByte(AK8963_ADDRESS, AK8963_ST1) & 0x01) {
                                         ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:249:15: note: declared here
       uint8_t readByte(uint8_t, uint8_t)   __attribute__ ((deprecated));
               ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:85:5: warning: 'void MPU9250::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*)' is deprecated [-Wdeprecated-declarations]
     readBytes(AK8963_ADDRESS, AK8963_XOUT_L, 7, &rawData[0]);
     ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:250:12: note: declared here
       void readBytes(uint8_t, uint8_t, uint8_t, uint8_t *)   __attribute__ ((deprecated));
            ^
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:85:60: warning: 'void MPU9250::readBytes(uint8_t, uint8_t, uint8_t, uint8_t*)' is deprecated [-Wdeprecated-declarations]
     readBytes(AK8963_ADDRESS, AK8963_XOUT_L, 7, &rawData[0]);
                                                            ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:250:12: note: declared here
       void readBytes(uint8_t, uint8_t, uint8_t, uint8_t *)   __attribute__ ((deprecated));
[... cut ...]
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:413:59: warning: 'uint8_t MPU9250::readByte(uint8_t, uint8_t)' is deprecated [-Wdeprecated-declarations]
   selfTest[5] = readByte(MPU9250_ADDRESS, SELF_TEST_Z_GYRO);  // Z-axis gyro self-test results
                                                           ^
In file included from /Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.cpp:2:0:
/Users/rozzie/Documents/Arduino/libraries/M5Stack-master/src/utility/MPU9250.h:249:15: note: declared here
       uint8_t readByte(uint8_t, uint8_t)   __attribute__ ((deprecated));
               ^

EeeeBin, 2019-07-12:
Try update your M5stack lib, MPU9250 Warnings already fixed in  https://github.com/m5stack/M5Stack/commit/102fd6655418931e8cd98bbe7fc93d363f956b99
LoRaWAN fixed in https://github.com/m5stack/M5Stack/commit/34bdefd119c11f2501437d87e04c3342a1caadb4

closed 2019-07-12 by EeeeBin

rayozzie, 2019-07-12:
I tried 0.2.8 (haven't tested your lorawan fixes yet).  The only non-Lorawan error is now this one:

/Users/rozzie/Documents/Arduino/libraries/M5Stack/src/M5Display.cpp: In member function 'void M5Display::drawBmpFile(fs::FS&, const char*, uint16_t, uint16_t)':
/Users/rozzie/Documents/Arduino/libraries/M5Stack/src/M5Display.cpp:127:23: warning: unused variable 'col' [-Wunused-variable]

Thanks for taking care of all this.