m5stack/M5Stack #186: Stepper motor module not reporting settings

bsculley, 2019-08-29:
The stepper motor module provides an I2C interface to GRBL.  The '$$' command is supposed to return all the settings, but all I am getting is:

$$

$0mnoing seek, mm/min)
$21=100 (homing debounce, msec)
$22=1.000 (homing pull-off, mm)
ok

This does not include the steps/mm ($100-$102) settings without which the module is useless.

Maybe I have a module with  downlevel firmware, but I have no idea how to upgrade it, or even if it's possible.  Any help will be appreciated.

Bob

EeeeBin, 2019-08-30:
step/mm not $100-$102, try test this https://github.com/m5stack/stepmotor_module/blob/master/Firmware%20for%20stepmotor%20module/GRBL-Arduino-Library/report.cpp#L146

bsculley, 2019-08-30:
Ok, I see from the code that X, Y and Z steps/mm  are $0, $1 and $2, but when I send $$, I am not getting the list indicated in the code.

 

Only this:

 

$$

 

$0mnoing seek, mm/min)

$21=100 (homing debounce, msec)

$22=1.000 (homing pull-off, mm)

ok

Apparently the code running on the device is not the same as what is shown in the link.  Is there a way to reflash the stepper control module?

 

Thanks

EeeeBin, 2019-08-30:
Tx buf not enough, so some data is lost.
https://github.com/m5stack/stepmotor_module/blob/b09cfa1c261a34c6aba9f3d4a904382dfb001fb2/Firmware%20for%20stepmotor%20module/GRBL-Arduino-Library/serial.h#L34

bsculley, 2019-08-30:
Ok, thanks.  But the code referenced is a part of GRBL, not part of the Arduino sketch.  

 

How can I apply any changes to the stepper controller module?

 

Is there a procedure to reflash the stepper controller?

 

https://github.com/m5stack/stepmotor_module/blob/b09cfa1c261a34c6aba9f3d4a904382dfb001fb2/Firmware%20for%20stepmotor%20module/GRBL-Arduino-Library/serial.h#L34

EeeeBin, 2019-08-30:
sorry, reflash step controllers is not yet supported

bsculley, 2019-08-30:
So how can I make it work?

Zontex, 2020-11-12:
Hello, what changes do you want to apply? if you want to change the steps you can solder the jumpers on the board to change the step modes (there are 3 different modes)