m5stack/M5Stack #190: MPU6886 data is not updated on M5Stack Gray

WindVoiceVox, 2019-09-21:
I tried with the source code below on M5Stack Gray.
The compilation is successful and it works, but the sensor information is not updated.

#include <M5Stack.h>
#include "utility/MPU6886.h"

MPU6886 IMU;

void setup() {
  // put your setup code here, to run once:
  int re;

  M5.begin();

  M5.Lcd.setBrightness(100);
  M5.Lcd.setTextSize(2);
  M5.Lcd.setTextColor(WHITE);

  re = IMU.Init();
}

float ax, ay, az;
float gx, gy, gz;
float t;

void loop() {
  // put your main code here, to run repeatedly:
  IMU.getAccelData(&ax, &ay, &az);
  IMU.getGyroData(&gx, &gy, &gz);
  IMU.getTempData(&t);

  M5.Lcd.fillScreen(BLACK);
  M5.Lcd.setCursor(0, 10);

  M5.Lcd.printf("Acc\n%.2f, %.2f, %.2f\nGyr\n%.2f, %.2f, %.2f\nTmp\n%.2f", ax, ay, az, gx, gy, gz, t);
  delay(100);
}

EeeeBin, 2019-09-21:
Try test this example https://github.com/m5stack/M5Stack/blob/master/examples/Basics/IMU/IMU.ino

WindVoiceVox, 2019-09-21:
Thanks, it works !

closed 2019-09-21 by WindVoiceVox