m5stack/M5Stack #191: M5Stack Fire MPU6886 and BMM150

Mystereon, 2019-10-10:
What's going on with this ?? The example for the Fire is for another MPU .. Please adapt this or provide the option as an example . Pretty please?  Or can anyone shed light ?

EeeeBin, 2019-10-11:
Pls try this https://github.com/m5stack/M5Stack/tree/master/examples/Basics/IMU

Mystereon, 2019-10-14:
Have done .. it doesn't mean much . the data is all over the place

radu7, 2019-10-28:
It seems the latest M5Stack Gray (MPU6886 + BMM150) comes loaded with the proper example 'FactoryTest.ino' that queries the device for either the MPU6866 or the MPU9250, but the example offered in this GitHub repo is only for the older MPU9250-based device...  Will the examples be updated to reflect what is currently shipping on the M5Stack core devices?

Mystereon, 2019-10-30:
If I attach the readings to M5.Lcd.Drawrect (pitch ,roll ,yaw, yaw, WHITE); ... the pitch and roll work to give me x and y ok, , but the yaw is ever increasing until it maxes and goes to zero again so i get an expanding box , irrespective of how I actually yaw the device.  I think mine is broken

radu7, 2019-10-30:
Mine actually behaves similarly.  At least when I load the 'IMU' example on my M5Stack Gray unit the yaw reading actually continuously gets smaller and smaller and cycles continuously that way, but the unit is sitting still on a flat/level surface.  If I pick it up and move it around or shake it a bit and if, at that point, I hold it at a 45 degree angle or more to the right (LCD facing up, buttons closest toward me) it might hold a bit, but eventually starts wandering up or down similar to what you describe.

I recall from my searching seeing at least one other person having the same issue with unstable yaw readings.  I believe they were attempting to use the M5Stack to detect if the unit had moved based on the yaw reading and were not able to do so due to this issue.

chaco71, 2019-11-05:
I'm having exactly the same problem with my recently purchased Fire, running the IMU sketch example (for the MPU6886): with the unit flat on the table, the yaw value steadily decreases until it rolls over. I can make it hold steady if the unit is inclined at about a 60 degree angle. Can this be fixed via a calibration routine?

So, the unit is pretty useless for my purposes, unless I can solve this issue. Has anyone found any solution? 

I was considering using this for a course I teach on interactive media, but I'm having second thoughts based on the lack of response on this important issue from the developers.

EeeeBin, 2019-11-06:
sorry, m5 only support an example show mpu6886 is working, if you want got  an accurate yaw value requires you to write your own integral algorithm

Mystereon, 2019-11-06:
How can you get a static yaw algorithm value from rotating yaw data ?
Let's see it work then?  only the x an y are implemented in the factory sketch.
I presume the digital compass is driving the yaw..

chaco71, 2019-11-06:
EeeeBin -- 
Your message is very frustrating to read. A lot of the interest in the M5Stack units is based on the IMU. I'm surprised M5Stack does not have a working example. A lot of your customers will lose faith in trying to figure this out if there is no guidance from the company. At the very least there should be accurate documentation and a path explained for how users may get the full features of the IMU working. Do you have suggestions for an integral algorithm to use with the MPU6886? (I know the MPU9250 had several working examples.)

AllenChao2019, 2019-11-07:
Hi Joseph & All,

Sorry for the confusion of IMU using in M5Stack Gray/GO/Fire. I need to explain the story for this double change. We used to use MPU9250, which is stable and good. But it was EOL for a long while. We have to switch to a new one. Then SH200Q came up. The headache thing is that, SH200Q is EOL again without advanced notice. So we have to switch to MPU6886. The latest one you have in hand should be MPU6886. We have posted this change in our Docs page, https://docs.m5stack.com/#/en/core/fire.

In regards to the unstable yaw data of 'IMU' example, that was caused by unstable 'Z' axis. Adding a GEOMAGNETIC sensor would enhance the accuracy. We have BMM150 inside. Just not adding an 'MPU6886+BMM150' example before, which would be released soon. (you can also try it with yourself if urgent)

P.S.: The data of BMM150 would be affected a lot by magnet. As there are magnet inside Core base. The data would not be in an accurate level with base. Thx!

Mystereon, 2019-11-07:
Aaaah , Allen , Thank you , Always it must be darkest before the dawn ... It makes sense that the magnetic base could affect the BMM150 . I think I speak for us all when I say I am keen to see the upcoming example you've hinted at.  Hold yer horses Chaco ,..  Dawn is coming!! :)

chaco71, 2019-11-13:
Hi Allen,

Thank you for your info about the IMU changes. I, too, would be very interested in seeing the MPU6886+BMM150 example you mention. I am involved in a project which is, in fact, somewhat urgent. Do you know when that example might be available?

Also, I see the BMM150 files in this directory on GitHub:
https://github.com/m5stack/M5StickC/tree/master/examples/Hat/ENV

Are those BMM150 components installed into the Arduino environment as part of the M5Stack library? Can I access the BMM150 directly?
Thanks.

chaco71, 2019-11-20:
Any responses to the question above, about accessing the BMM150?

kleinholdermann, 2019-12-02:
Any news on this? Bought the device mainly because of the IMU. It's more or less useless as it is at the moment.

EeeeBin, 2019-12-03:
you can try use it: https://github.com/m5stack/M5Stack/tree/master/examples/Basics/bmm150

twinotter, 2019-12-27:
This is a great start, but could you expand this to include the 6886 .. and ideally filtering functions?

Thanks!

omegatao, 2020-01-19:
I have writen an example on how to calculate AHRS parameters using M5Stack grey with MPU6886 and BMM150 sensor, you can check it out. [M5StackGrey_MPU6886_BMM150_AHRS_sample](https://github.com/omegatao/M5StackGrey_MPU6886_BMM150_AHRS_sample)     https://github.com/omegatao/M5StackGrey_MPU6886_BMM150_AHRS_sample

chaco71, 2020-01-19:
Great — looking forward to checking it out!

> I have writen an example on how to calculate AHRS parameters using M5Stack grey with MPU6886 and BMM150 sensor, you can check it out. M5StackGrey_MPU6886_BMM150_AHRS_sample [...]

------------------------------------------------------
Joseph Butch Rovan
Professor of Music
Brown University
1 Young Orchard Ave., Box 1924
Providence, RI 02912 USA 

O: 401.863.1793 M: 401.699.2759 
rovan@brown.edu 
soundidea.org

cyberluke, 2020-01-23:
There is no Invensense / TDK datasheet for MPU-6886. Seems it does not exist! Also, it is not on their website. I found only a datasheet for 6880, but it is from 2013. Even older than 9250.

Why you use this?

The equivalent for 9250 is ICM-20948:
https://www.cdiweb.com/datasheets/invensense/an-000146-v2-tdk-migration-mpu-9250-to-icm-20948.pdf

Why, why & why?

MartyMacGyver, 2020-02-12:
> There is no Invensense / TDK datasheet for MPU-6886. Seems it does not exist! Also, it is not on their website. I found only a datasheet for 6880, but it is from 2013. Even older than 9250.

Perhaps this will help...

https://github.com/m5stack/M5-Schematic/blob/master/datasheet/MPU-6886-000193%2Bv1.1_GHIC.PDF.pdf

cyberluke, 2020-02-12:
Thank you, I am a registered Invensense Developer and they don't have it. Still, for new designs, there is ICM-20948 already for one or two years. It has better SDK.

valki2, 2020-06-08:
Any news on this topic? Seems that there is still no working MPU6886+BMM150 example around while the MPU6886 yaw is just increasing nonsense.

ang146, 2020-08-20:
> I have writen an example on how to calculate AHRS parameters using M5Stack grey with MPU6886 and BMM150 sensor, you can check it out. [...]

I tried this sample. The yaw value is just like being blocked in some "area" and the value just jumped over after that "area"

uecken, 2020-12-06:
> I tried this sample. The yaw value is just like being blocked in some "area" and the value just jumped over after that "area"

I also tried this code, however Yaw will be around 130 with the display facing up.
https://drive.google.com/file/d/1TAjT1psyprSbvtIQevGWK4aV5P12AkCe/view?usp=sharing

The M5Stack-Gray have a little magnetic, is it the cause for bad yaw?

omegatao, 2020-12-07:
> I also tried this code, however Yaw will be around 130 with the display facing up.
> https://drive.google.com/file/d/1TAjT1psyprSbvtIQevGWK4aV5P12AkCe/view?usp=sharing
>
> The M5Stack-Gray have a little [...]

I have also experienced some issue with the inaccurate readings. I implemented those code by the manual, so I think that these problems maybe caused by poor calibration or electromagnetic interference.

9PEO0xNd, 2021-01-21:
Hi! I have tweaked MxMyMz and added soft iron calibration in omegatao's library. If you are using arduino IDE, try https://github.com/9PEO0xNd/M5Gray_MPU6886_BMM150

omegatao, 2021-01-23:
> Hi! I have tweaked MxMyMz and added soft iron calibration in omegatao's library. If you are using arduino IDE, try https://github.com/9PEO0xNd/M5Gray_MPU6886_BMM150

cool, you can also create a pull request to the platform IO based code if you like, in case of someone just wants to use the platform IO version.

chaco71, 2021-01-23:
Thanks very much — looking forward to trying it out!

> Hi! I have tweaked MxMyMz and added soft iron calibration in omegatao's library. If you are using arduino IDE, try https://github.com/9PEO0xNd/M5Gray_MPU6886_BMM150 [...]

------------------------------------------------------
Joseph Butch Rovan
Professor of Music
Brown University
1 Young Orchard Ave., Box 1924
Providence, RI 02912 USA 

O: 401.863.1793 M: 401.699.2759 
rovan@brown.edu 
soundidea.org

grundprinzip, 2021-01-28:
Using https://github.com/9PEO0xNd/M5Gray_MPU6886_BMM150 I was able to get reliable yaw, pitch, and roll readings compared to the example provided directly from M5Stack.

AllenChao2019, 2021-03-30:
> Hi! I have tweaked MxMyMz and added soft iron calibration in omegatao's library. If you are using arduino IDE, try https://github.com/9PEO0xNd/M5Gray_MPU6886_BMM150

Thx for sharing, 9PEO0xNd. That is great!

edumo, 2021-05-03:
Working example for m5stickc https://github.com/edumo/M5StickC_Imu-9-axis

noxqs, 2022-11-29:
To come back to the start of this discussion, did anyone got this to work on the Fire ?