m5stack/M5Stack #193: ENOMEM when making a http request

DonJayamanne, 2019-11-23:
Got the M5Stack grey (4MB of FLASH), and it's failing with an ENOMEM error when making an http request.

Itsa simple program uploaded using UIFlow.
Fails as soon as the http request is made using urequests

EeeeBin, 2019-11-23:
ESP32 memory not enough to got all http request
Pls try got some web page with low content

DonJayamanne, 2019-11-23:
Doesn't the M5Stack have 4MB?

DonJayamanne, 2019-11-23:
I take that back, you are right. I was a few KB off...lol, hit a limit there.
Thanks and apologies

DonJayamanne, 2019-11-23:
Sorry, I take that back, i was swallowing the exceptions.

EeeeBin, 2019-11-23:
Sorry, I test http request got 1KB file successful (1.4.2),  could not reply this problem
Can you provide a more detailed log?
And this is arduino repositories, so this issue may be better open in here --> https://github.com/m5stack/UiFlow-Bug-Report/issues

closed 2019-11-23 by EeeeBin