m5stack/M5Stack #194: MPU9250BasicAHRS Example Guru Meditation Error

sandscameron, 2019-11-27:
Hoping I may be able to get some help with the error I get in the MPU925BasicAHRS example. If I remove the line for the calibration of MPU9250, the rest works. I am running it on an m5stack Gray. The line in question is:
IMU.calibrateMPU9250(IMU.gyroBias, IMU.accelBias);

At the point this is called, I get the following error (in the serial monitor) with it then going into a reboot:
Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d49e6  PS      : 0x00060330  A0      : 0x800d193a  A1      : 0x3ffb1f10  
A2      : 0x3ffb1f40  A3      : 0x3ffc01f0  A4      : 0x3ffc01fc  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000002  A12     : 0x00000000  A13     : 0x00000002  
A14     : 0x3ffb1f40  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400d49e6:0x3ffb1f10 0x400d1937:0x3ffb1f70 0x400d9ef9:0x3ffb1fb0 0x40088c19:0x3ffb1fd0

Rebooting...

Any direction to the error would be appreciated.

EeeeBin, 2019-11-28:
**IntegerDivideByZero** Is mean:  https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/fatal-errors.html#integerdividebyzero
Try this example: https://github.com/m5stack/M5Stack/tree/master/examples/Basics/IMU
If want know error detail, try use this tool: https://github.com/me-no-dev/EspExceptionDecoder

sandscameron, 2019-11-28:
I am happy with how the IMU example you gave works so have not followed through why the calibrate call of the other example gets a divide by zero.
Thanks for your reply.

closed 2019-11-29 by EeeeBin