m5stack/M5Stack #203: Platformio + M5Stack Lcd not working properly

analogic, 2020-03-15:
I am trying to get Lcd working but still hitting same problem - once I am start using Lcd it doesn't work with anything else than Lcd.print* functions.

Steps to reproduce
0. get VSCode, install platformio and M5Stack lib on Mac OS
1. copy default example https://platformio.org/lib/show/1851/M5Stack
2. compile and upload
3. nothing(or buffer from previous program) displayed

Strange thing is that in Arduino IDE everything works as intended. Another strange thing is that print function works as intended until there is some drawing function included in code then Lcd stops working without no error and other code works as usual. Might be some linking bug? Or *.ino vs *.cpp files handling bug? Is something wrong with my dev stack?

EeeeBin, 2020-03-16:
Sorry
After testing, it is found to be running well
May be a development environment issue

analogic, 2020-03-16:
Ok, closing for now. Will try to reinstall whole dev env

closed 2020-03-16 by analogic